platformio/platform-ststm32 #95: STM32Cube framework is listed twice for the BluePill

jcw, 2018-03-15:
As the title says. Not sure what's going on here.

ivankravets, 2018-03-15:
Could you provide a screenshot?

ubis, 2018-03-15:
I think @jcw meant this: [L41](https://github.com/platformio/platform-ststm32/blob/develop/boards/bluepill_f103c8.json#L41) and [L44](https://github.com/platformio/platform-ststm32/blob/develop/boards/bluepill_f103c8.json#L44).

jcw, 2018-03-15:
Ah, sorry, yes:

[image]

ivankravets, 2018-03-15:
Thanks, guys! I fixed.

@jcw Which IDE do you use? With which theme?

closed 2018-03-15 by ivankravets

jcw, 2018-03-15:
Atom w/ Atom Light on Mac. Thx for fixing. If I run into other little quirks, I'll submit more little issues :)

ivankravets, 2018-03-15:
Great! :)

Happy coding with PlatformIO!

commit 29edb61b (2018-03-15): Remove duplicated stm32cube framework // Resolve #95