platformio/platform-ststm32 #98: Add mbed framework to STM32F103CB

maxgerhardt, 2018-03-17:
The STM32F103CB is the 128K flash version of the STM32F103C8 chip (64K flash) which also powers the bluepill board. (On [most bluepill boards](http://wiki.stm32duino.com/index.php?title=Blue_Pill#128_KB_flash_on_C8_version) you can actually access the whole 128K of flash as if it were a CB version.) So if the board `genericSTM32F103C8` supports mbed, the CB version with just the bigger flash should be supported, too.

It is also confusing that the board "bluepill_f103c8" (doesn't support mbed) is different from "genericSTM32F103C8" (supports mbed) although they are listed to have the exact same MCU.

I tried adding "mbed" to the list of frameworks myself in the boards/genericSTM32F103CB.json file which resulted in PIO complaining about a missing config file. After going to `.platformio\packages\framework-mbed\platformio` and following the instructions in the `README.txt` the command `tox` failed with error 

```
Current target UBLOX_EVA_NINA
Current target VBLUNO51_OTA
Traceback (most recent call last):
  File "piombedgenerator/main.py", line 268, in <module>
    sys_exit(main())
  File "piombedgenerator/main.py", line 204, in main
    toolchain.config.load_resources(framework_resources)
  File "..\tools\config\__init__.py", line 937, in load_resources
    features = set(self.get_features())
  File "..\tools\config\__init__.py", line 891, in get_features
    params, _ = self.get_config_data()
  File "..\tools\config\__init__.py", line 819, in get_config_data
    all_params = self.get_target_config_data()
  File "..\tools\config\__init__.py", line 765, in get_target_config_data
    ConfigParameter.get_display_name(tname, "target")))
tools.config.ConfigException: Attempt to override undefined parameter 'uart_hwfc' in 'target:VBLUNO51_OTA'
ERROR: InvocationError: 'C:\\Users\\Maxi\\.platformio\\packages\\framework-mbed\\platformio\\.tox\\python\\Scripts\\python.EXE piombedgenerator/main.py'
```
so I hit a road-block there.

ivankravets, 2018-07-10:
Could someone test it?

1) Edit `~/.platformio/platforms/ststm32/boards/genericSTM32F103CB.json` and add "mbed" to `"frameworks"` list
2) Edit `~/.platformio/packages/framework-mbed/platformio/variants_remap.json` and add 
`"genericSTM32F103CB": "BLUEPILL_F103C8"`

Does it work?

filipposanfilippo, 2018-08-17:
ivankravets, I have tried your suggestion but without success. Any ideas?

> Error: Please specify `upload_port` for environment or use global `--upload-port` option.

ivankravets, 2018-08-17:
@filipposanfilippo do you use `upload_protocol = stlink`? What is your programmer?

maxgerhardt, 2018-11-04:
@ivankravets I've tried your suggestion and it does compile normal firmwares, but the used / generated linker (LD) file is still wrong. Compiling a project results in a `pioenvs\<envname>\STM32F103XB.ld.link_script.ld` file which still has 

```MEMORY
{
  FLASH (rx) : ORIGIN = 0x08000000, LENGTH = 64K
  RAM (rwx) : ORIGIN = 0x200000EC, LENGTH = 20K - 0xEC
}
```
Thus the correct flash value of 128K isn't used. If you edit this file 128K and then reupload after an error occurs (could not fit .. in region FLASH, overrun by XY bytes) it works.

ivankravets, 2018-11-06:
@valeros please help here.

valeros, 2018-11-06:
At the moment I see two ways how to handle this issue:
1. Create a custom linker script:
a) Add a new variant `"genericSTM32F103CB": "BLUEPILL_F103C8"` to `variants_remap.json` file. 
b) Build your project once, copy generated linker script from build folder (`.pioenvs/env/STM32F103XB.ld.link_script.ld`) to the root of your project and give it a specific name (e.g. STM32F103XB_128KB.ld).
c) Edit flash size in the new ld script and specify it in `platformio.ini`
`build_flags = -Wl,-TSTM32F103XB_128KB.ld.ld`
2. Create a fully functional variant for the F103CB target (all paths below are realtive to the root folder of the frameowrk):
a) Create a copy of the next folder `platformio\variants\BLUEPILL_F103C8` and rename it to `F103CB` (rename `BLUEPILL_F103C8.json` to `F103CB.json` as well)
b) Open file `F103CB.json`, find the field  `ldscript` and edit its value to  `targets/TARGET_STM/TARGET_STM32F1/TARGET_BLUEPILL_F103C8/device/TOOLCHAIN_GCC_ARM/STM32F103XB_128KB.ld`,
c) Go to the folder `targets/TARGET_STM/TARGET_STM32F1/TARGET_BLUEPILL_F103C8/device/TOOLCHAIN_GCC_ARM/`, copy file `STM32F103XB.ld`, rename it to `STM32F103XB_128KB.ld`, specify required flash size.
d) Open `platformio/variants_remap.json` and add next line `"genericSTM32F103CB": "F103CB"`

Hope this helps.

Sambo007, 2018-11-28:
One more thing is missing from @valeros steps:

add "mbed", to ~\.platformio\platforms\ststm32\boards\genericSTM32F103C8.json as follows:
```
  "frameworks": [
    "arduino",
    "mbed",
    "libopencm3",
    "stm32cube"
  ],
```
Otherwise, this error will pop up:
```
Error: This board doesn't support mbed framework!
```
Which would make more sense if it says:
```
Error: This board is not supported in mbed framework!
```

Sambo007, 2018-11-28:
Using 'blue pill' with STM32F103C8.

Building worked OK:
```
> Executing task: C:\Users\Samir\.platformio\penv\Scripts\platformio.exe run <

Processing genericSTM32F103CB (framework: mbed; platform: ststm32; board: genericSTM32F103CB)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103CB.html
PLATFORM: ST STM32 > STM32F103CB (20k RAM. 128k Flash)
HARDWARE: STM32F103CBT6 72MHz 20KB RAM (128KB Flash)
DEBUG: CURRENT(stlink) EXTERNAL(blackmagic, jlink, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 43 compatible libraries
Scanning dependencies...
.
.
.
Linking .pioenvs\genericSTM32F103CB\firmware.elf
Building .pioenvs\genericSTM32F103CB\firmware.bin
Checking size .pioenvs\genericSTM32F103CB\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  28.8% (used 5896 bytes from 20480 bytes)
PROGRAM: [=====     ]  50.8% (used 66524 bytes from 131072 bytes)
================================================================== [SUCCESS] Took 17.06 seconds ==================================================================
```

However, Uploading did not work:
```
Linking .pioenvs\genericSTM32F103CB\firmware.elf
Checking size .pioenvs\genericSTM32F103CB\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  28.8% (used 5896 bytes from 20480 bytes)
PROGRAM: [=====     ]  50.8% (used 66524 bytes from 131072 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, dfu, jlink, serial, stlink
CURRENT: upload_protocol = stlink
Uploading .pioenvs\genericSTM32F103CB\firmware.elf
GNU MCU Eclipse 64-bits Open On-Chip Debugger 0.10.0+dev-00392-gbe9ef0b0 (2018-01-12-15:03)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v32 API v2 SWIM v7 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.328000
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08000a44 msp: 0x20005000
** Programming Started **
auto erase enabled
Info : device id = 0x20036410
Info : flash size = 64kbytes
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000003a msp: 0x20005000
Warn : no flash bank found for address 8010000
wrote 65536 bytes from file .pioenvs\genericSTM32F103CB\firmware.elf in 2.051117s (31.203 KiB/s)
** Programming Finished **
** Verify Started **
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000002e msp: 0x20005000
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000002e msp: 0x20005000
Error: checksum mismatch - attempting binary compare
diff 0 address 0x08010000. Was 0xff instead of 0x30
diff 1 address 0x08010001. Was 0xff instead of 0x09
diff 2 address 0x08010002. Was 0xff instead of 0x00
.
.
.
diff 127 address 0x0801007f. Was 0xff instead of 0x20
More than 128 errors, the rest are not printed.
embedded:startup.tcl:476: Error: ** Verify Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 520
at file "embedded:startup.tcl", line 476
*** [upload] Error 1
=================================================================== [ERROR] Took 23.85 seconds ===================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```
ST-LINK still sees 64kB only!

Note that I was able to flash the program, which is more than 64KB, using 'J-Flash V6.40' and it worked fine.

maxgerhardt, 2018-11-28:
>ST-LINK still sees 64kB only! 

Does the ST-LINK Utility see 128KB or 64KB? Not all blue pills boards actually are 128K.

Sambo007, 2018-11-28:
STM32 ST-LINK Utility v4.3.0.0 reads CPU ID and will not flash the C8 beyond 64K. I didn't try older/different ST-LINK utility that others have been able to use to program the C8 as CB.

I already flashed a program >64K with J-Flash V6.40 and it verified OK. The program is also running fine with USB Serial. So that answers that my blue pill is 128K.

I also managed to get upload working w/ J-LINK (see upload dump below) using the following
```
upload_protocol = jlink
extra_scripts   = extra_script.py
```
But I have to build first then upload. Otherwise, the firmware.bin is not generated by clicking upload alone.
The extra_script.py is necessary since upload did not work using upstream (`platform = https://github.com/platformio/platform-ststm32.git`) as it did for `framework = Arduino` without  the script.

Finally, I hope STM32F103CB gets added to mbed framework and the necessary changes to make jlink upload/debug in the next stable release ASAP. I am not sure how can the current ST-LINK Utility be tricked to see C8 as CB. Maybe we should revert to a different/older ST utility.

```
> Executing task: C:\Users\Samir\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing genericSTM32F103CB (framework: mbed; platform: ststm32; board: genericSTM32F103CB)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103CB.html
PLATFORM: ST STM32 > STM32F103CB (20k RAM. 128k Flash)
HARDWARE: STM32F103CBT6 72MHz 20KB RAM (128KB Flash)
DEBUG: CURRENT(jlink) EXTERNAL(blackmagic, jlink, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 43 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <USBSerial>
|   |-- <USBDevice>
Checking size .pioenvs\genericSTM32F103CB\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  28.8% (used 5896 bytes from 20480 bytes)
PROGRAM: [=====     ]  50.8% (used 66548 bytes from 131072 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, dfu, jlink, serial, stlink
CURRENT: upload_protocol = jlink
Uploading .pioenvs\genericSTM32F103CB\firmware.elf
SEGGER J-Link Commander V6.40 (Compiled Oct 26 2018 15:06:29)
DLL version V6.40, compiled Oct 26 2018 15:06:02

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Firmware: J-Link ARM V8 compiled Nov 28 2014 13:44:46
Hardware version: V8.00
S/N: -1
License(s): RDI,FlashDL,FlashBP,JFlash,GDBF
VTref=3.358V
Target connection not established yet but required for command.
Device "STM32F103CB" selected.

Connecting to target via SWD
Found SW-DP with ID 0x1BA01477
Found SW-DP with ID 0x1BA01477
Scanning AP map to find all available APs
AP[1]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x14770011)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x411FC231. Implementer code: 0x41 (ARM)
Found Cortex-M3 r1p1, Little endian.
FPUnit: 6 code (BP) slots and 2 literal slots
CoreSight components:
ROMTbl[0] @ E00FF000
ROMTbl[0][0]: E000E000, CID: B105E00D, PID: 001BB000 SCS
ROMTbl[0][1]: E0001000, CID: B105E00D, PID: 001BB002 DWT
ROMTbl[0][2]: E0002000, CID: B105E00D, PID: 000BB003 FPB
ROMTbl[0][3]: E0000000, CID: B105E00D, PID: 001BB001 ITM
ROMTbl[0][4]: E0040000, CID: B105900D, PID: 001BB923 TPIU-Lite
Cortex-M3 identified.
PC = 2000054C, CycleCnt = 0004E355
R0 = 40003000, R1 = 0000AAAA, R2 = 20000714, R3 = 00000004
R4 = 200006BC, R5 = 200006DC, R6 = 200006DC, R7 = 3F800000
R8 = 20000DA0, R9 = FFFDFFDC, R10= 1FFF5000, R11= 00000000
R12= 00000000
SP(R13)= 200006A0, MSP= 200006A0, PSP= 283FA33C, R14(LR) = 2000032F
XPSR = 21000000: APSR = nzCvq, EPSR = 01000000, IPSR = 000 (NoException)
CFBP = 00000000, CONTROL = 00, FAULTMASK = 00, BASEPRI = 00, PRIMASK = 00
FPU regs: FPU not enabled / not implemented on connected CPU.

Downloading file [d:\Embedded\ST\STM32F103\Examples\VSCode\Mbed\RTC\01_Time\.pioenvs\genericSTM32F103CB/firmware.bin]...
J-Link: Flash download: Bank 0 @ 0x08000000: 1 range affected (66560 bytes)
J-Link: Flash download: Total time needed: 2.064s (Prepare: 0.058s, Compare: 0.026s, Erase: 0.000s, Program: 1.964s, Verify: 0.009s, Restore: 0.004s)
O.K.

Reset delay: 0 ms
Reset type NORMAL: Resets core & peripherals via SYSRESETREQ & VECTRESET bit.
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.

Script processing completed.

================================================================== [SUCCESS] Took 10.45 seconds ==================================================================
```

Sambo007, 2018-11-30:
I think this might help fix the 64k/128k issue using openocd:
[#013: Unlock the Hidden Flash of STM32F103C8](https://www.youtube.com/watch?v=0eHpoPZvI3U)

harimambura, 2018-11-30:
@ivankravets Recently you closed issue about generic STM32F103RE, and what about this board? in targets folder I found only generic board, that don't support mbed framework, must I repeat the steps, that was provided by valeros?

harimambura, 2018-11-30:
I tried to repeat the steps, but after the second variant (create new variant folder end etc) I had an error ".platformio\packages\framework-mbed\targets\TARGET_STM\TARGET_STM32F1\TARGET_NUCLEO_F103RB\device/stm32f1xx.h:151:12: fatal error: stm32f103xe.h: No such file or directory" during building project

maxgerhardt, 2019-02-08:
128K flash version was added with [this PlatformIO update](https://community.platformio.org/t/arm-mbed-os-5-11-new-builder-with-support-for-mbed-app-json/6471), closing; thanks @ivankravets @valeros 👍

closed 2019-02-08 by maxgerhardt

yozik04, 2019-10-08:
There is still no board with mbed framework and 128k in
https://platformio.org:443/boards?count=15&filter%5Bframeworks%5D=mbed&filter%5Bmcu%5D=STM32F103C8&page=1&sorting%5Bname%5D=asc