platformio/platform-ststm32 #99: Blue Pill(STM32F103C8) not work with stm32cube

nmhttt, 2018-03-21:
I have Blue Pill board which use STM32F103C8, it is ok with arduino or mbed framework but not with stm32cube. It  sucess complie and upload but the led don't blink. I think there are some problems with linkder script ?

There is my config

```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = stm32cube
```

```
#include "stm32f1xx_hal.h"

#define LED_PIN GPIO_PIN_13
#define LED_GPIO_PORT GPIOC
#define LED_GPIO_CLK_ENABLE() __HAL_RCC_GPIOA_CLK_ENABLE()

void LED_Init();

int main(void)
{
    HAL_Init();
    LED_Init();

    while (1)
    {
        HAL_GPIO_TogglePin(LED_GPIO_PORT, LED_PIN);
        HAL_Delay(500);
    }
}

void LED_Init()
{
    LED_GPIO_CLK_ENABLE();
    GPIO_InitTypeDef GPIO_InitStruct;
    GPIO_InitStruct.Pin = LED_PIN;
    GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
    GPIO_InitStruct.Pull = GPIO_PULLUP;
    GPIO_InitStruct.Speed = GPIO_SPEED_HIGH;
    HAL_GPIO_Init(LED_GPIO_PORT, &GPIO_InitStruct);
}
```

maxgerhardt, 2018-03-23:
Your LED port is on GPIOC but you enable the clock for GPIOA? Has the code worked in STM32Cube itself without PIO?

nmhttt, 2018-03-24:
i have fixed, but still now work

nmhttt, 2018-03-24:
i found problem is with HAL_DELAY() function.

maxgerhardt, 2018-03-24:
This is strictly a code problem and no PlatformIO problem.  You should start with a STM32Cube project and import it into PIO, then modify it. You missing several thigs: In order for `HAL_Delay()` to work, you must hook the Systick handler and call `HAL_IncTick()`. You are also missing the intialiation of the PLL to set your CPU clock to 72 MHz.

This code is working properly on my Bluepill board.

```cpp
#include "stm32f1xx_hal.h"
#include "stm32f1xx.h"

#define LED_PIN GPIO_PIN_13
#define LED_GPIO_PORT GPIOC
#define LED_GPIO_CLK_ENABLE() __HAL_RCC_GPIOC_CLK_ENABLE()

void _Error_Handler(const char* file, int line) {
	//Error occured.
}

/* Boot up the Phase Locked Loop to give us a 72MHz clock.. */
void SystemClock_Config(void)
{

  RCC_OscInitTypeDef RCC_OscInitStruct;
  RCC_ClkInitTypeDef RCC_ClkInitStruct;

    /**Initializes the CPU, AHB and APB busses clocks
    */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSE;
  RCC_OscInitStruct.HSEState = RCC_HSE_ON;
  RCC_OscInitStruct.HSEPredivValue = RCC_HSE_PREDIV_DIV1;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSE;
  RCC_OscInitStruct.PLL.PLLMUL = RCC_PLL_MUL9;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    _Error_Handler(__FILE__, __LINE__);
  }

    /**Initializes the CPU, AHB and APB busses clocks
    */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV2;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_2) != HAL_OK)
  {
    _Error_Handler(__FILE__, __LINE__);
  }

    /**Configure the Systick interrupt time
    */
  HAL_SYSTICK_Config(HAL_RCC_GetHCLKFreq()/1000);

    /**Configure the Systick
    */
  HAL_SYSTICK_CLKSourceConfig(SYSTICK_CLKSOURCE_HCLK);

  /* SysTick_IRQn interrupt configuration */
  HAL_NVIC_SetPriority(SysTick_IRQn, 15, 0);
}

/* We must hook the systick in order for HAL_Delay to work. */
extern "C" void SysTick_Handler(void)
{
	HAL_IncTick();
  //osSystickHandler(); For FREERTOS stuff
}

void LED_Init()
{
    LED_GPIO_CLK_ENABLE();
    GPIO_InitTypeDef GPIO_InitStruct;
    GPIO_InitStruct.Pin = LED_PIN;
    GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    GPIO_InitStruct.Speed = GPIO_SPEED_HIGH;
    HAL_GPIO_Init(LED_GPIO_PORT, &GPIO_InitStruct);
}

int main() {

	//Init HAL
	HAL_Init();

	//Initialize clocks to max speed
	SystemClock_Config();

	//Init GPIOs
	LED_Init();

	//HAL_GPIO_WritePin(GPIOC, GPIO_PIN_13, GPIO_PIN_RESET);

	//Blinky blinky
	while(1) {
		HAL_GPIO_TogglePin(LED_GPIO_PORT, LED_PIN);
		HAL_Delay(1000);
	}

	return 0;
}
```
If it works for you you can close the issue.

closed 2018-03-24 by nmhttt

brunobasto, 2021-06-29:
In case someone finds this in present days:

Had the same problem, all it took to solve was adding the `extern "C"` to `SysTick_Handler`:

```cpp
extern "C" void SysTick_Handler(void)
{
	HAL_IncTick();
  //osSystickHandler(); For FREERTOS stuff
}
```

complete code

```cpp
#include "stm32f1xx_hal.h"

#define LED_PIN GPIO_PIN_12
#define LED_GPIO_PORT GPIOB
#define LED_GPIO_CLK_ENABLE() __HAL_RCC_GPIOB_CLK_ENABLE()

int main(void)
{
    HAL_Init();

    LED_GPIO_CLK_ENABLE();

    GPIO_InitTypeDef GPIO_InitStruct;

    GPIO_InitStruct.Pin = LED_PIN;
    GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
    GPIO_InitStruct.Pull = GPIO_PULLUP;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH;
    HAL_GPIO_Init(LED_GPIO_PORT, &GPIO_InitStruct);

    while (1)
    {
        HAL_GPIO_TogglePin(LED_GPIO_PORT, LED_PIN);
        HAL_Delay(1000);
    }
}

extern "C" void SysTick_Handler(void)
{
	HAL_IncTick();
 }
```