platformio/platform-ststm32 #102: Linker error with "maple_mini_b20" board

jcw, 2018-04-04:
Hello - I've set up a new project with the following Maple settings for an Olimexino-STM32:

```
[env:maple]
platform = ststm32
board = maple
framework = arduino
```

It builds and uploads fine (I installed the default bootloader, from the first link in the SOFTWARE section of the [Olimexino](https://www.olimex.com/Products/Duino/STM32/OLIMEXINO-STM32/) home page). There's a serial USB console, and the board automatically enters boot mode when the PIO upload starts.

The uploaded code does not _start_ automatically though. You have to press the reset button for that.

In this setup, the sketch has 108 out of 128K flash and 17 out of 20K RAM. Plenty for me.

But I wanted to try the newer STM32duino 2.0 bootloader (i.e. `maple_mini_b20` iso `maple` board), mostly in the hope that automatic re-upload **and** restarting would work. It runs into this error, however:

```
[...]
arm-none-eabi-ranlib .pioenvs/maple/libFrameworkArduino.a
arm-none-eabi-g++ -o .pioenvs/maple/firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m3 -Wl,-T"bootloader_20" .pioenvs/maple/src/main.cpp.o -L/Users/jcw/.platformio/platforms/ststm32/ldscripts -L/Users/jcw/.platformio/packages/framework-arduinoststm32/STM32F1/variants/maple_mini/ld -L.pioenvs/maple -Wl,--start-group .pioenvs/maple/libFrameworkArduinoVariant.a .pioenvs/maple/libFrameworkArduino.a -lc -lgcc -lm -lc -Wl,--end-group
/Users/jcw/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld: cannot open linker script file bootloader_20: No such file or directory
collect2: error: ld returned 1 exit status
```

I found copies of this file inside `~/.platformio/packages/framework-arduinoststm32/STM32F1/variants`, but somehow they're not found / not in the right place?

ubis, 2018-04-04:
This is happening due missing `*.ld` extesion in [L77](https://github.com/platformio/platform-ststm32/blob/develop/builder/frameworks/arduino/maple/stm32f1.py#L77).
Though it's wierd, because `Maple` uses bootloader v1 and `Maple Mini Bootloader 2.0` uses v2.

BTW you could also use `genericSTM32F103RB` with `dfu` as upload protocol.

jcw, 2018-04-04:
> you could also use genericSTM32F103RB

Aha, thanks for the link to [stm32f1.py](https://github.com/platformio/platform-ststm32/blob/develop/builder/frameworks/arduino/maple/stm32f1.py), that clears up a lot w.r.t. how the different load offsets are used!

ubis, 2018-04-04:
Since you mentioned macOS, there is something wrong with it and dfu-util. Similar issue #37 
Perhaps could you try to write upload port manually with `upload_port=/dev/...`?

I would start troubleshooting with Arduino IDE first, in order to know if it's our problem or it's `dfu's` bug but unfortunately, I don't use macOS 🙁

jcw, 2018-04-04:
I can look into it, but I now think that last issue I mentioned has a different cause: the sketch is built for 0x08002000, and the Maple loader I have uses the lower 20K, so it can't load what I'm sending it.

I've yet to fully understand all the different boot mechanisms. As I understand it:

* the original Maple boot loader used 20K flash and 3K ram, and loaded coded above both
* the "updated" boot loader (r5?) still used 20K flash, but dropped the 3K ram requirement?
* the stm32duino v2 boot loader uses 8K flash, leaves all ram to the sketch, and loads to 0x08002000
* for backwards compatibility, that v2 boot loader can also load and launch a sketch at 0x08005000

Still looking around: lots of versions, lots of binaries, so little links back to the exact code used in each...

jcw, 2018-04-04:
> try to write upload port manually with `upload_port=/dev/...`?

I don't know whether that can be done. DFU doesn't talk to serial ports, so there's no `/dev/cu.*` for the boot loader, only for the serial USB port while a sketch is running. Maybe the uid/vid or serial or device path needs to be specified in `upload_flags`.

ubis, 2018-04-04:
I'm unsure about "r5 maple bootloader". As far as I know, there are two bootloaders: 1.0 and 2.0.
1.0 - original Maple bootloader, some ram are used and 0x08005000
2.0 - stm32duino bootloader, none of ram are used and 0x08002000

Now, all generic F103 boards, if used with `dfu` as upload protocol, will use 2.0.
Maple; Maple Mini Original - 1.0; Maple Mini Bootloader 2.0 - 2.0.

> DFU doesn't talk to serial ports, so there's no `/dev/cu.*` for the boot loader, only for the serial USB port while a sketch is running.

Yes, because dfu-util sends a reset signal to this serial port, after reset, dfu-util should start flashing.

> Maybe the uid/vid or serial or device path needs to be specified in `upload_flags`.

We have set VID/PID in board's configuration [file](https://github.com/platformio/platform-ststm32/blob/develop/boards/genericSTM32F103RB.json#L7), so there's no need for that.

jcw, 2018-04-04:
Oh, and just for completeness: I _never_ see an upload followed by a running sketch. I always have to press the reset button to start the just-uploaded code.

ubis, 2018-04-04:
That after upload seems like an USB issue. Is your board a clone? Could you check if USB pull-up resistor is 1k5? More [info](http://wiki.stm32duino.com/index.php?title=Blue_Pill) about this. I had similar problems, like after flashing dfu stopped working, until I replaced resistor.

However, reset-after-flash should work, unless it's macOS/dfu-util issue...

jcw, 2018-04-04:
There are also some timing issues - when I use the Black Magic Probe to re-flash (which does properly launch after upload), it will enter the DFU bootloader, which wil sit around for a few seconds. If I then also have a serial monitor open, that will fail, as the serial USB port will take a few seconds to appear.

There are many side-effects in all this. I wonder if there's a way to get rid of a few of these issues. Right now, F103's – being the most widespread and cloned ARM boards for hobbyists – are really tricky. Having several dozen different of those boards here, I'd be interested in helping improve things a bit, but to be honest, I don't quite know where to start. The fastest dev-cycle I found so far is a Black Magic Probe connected to USART1 for console/debugging (which gets it down to under 2s per compile/upload), but I'd love to find a good setup which works entirely over USB. The DTR + DFU approach is very nice, but there are sooo many things that can go wrong. I'm also surprised that DFU uploads of even just 15 KB firmware images take so long.

I'm not complaining. The range of boards, pull-ups, F103 variants, frameworks, etc is _staggering_, and it's probably very hard to make things simpler for people who just want to code and get on with it.

ubis, 2018-04-04:
These bootloader binaries can be used on any F103 MCU. `generic_boot20_pa1` means that LED is on PA1.
Even if it's a clone, it could have a wrong pull-up resistor, doesn't matter that the correct one is shown in schematics.

All these timing-issues are similar to USB-related issues, due wrong pull-up. Or I could be wrong and it would mean that it's macOS/dfu-util issue. There are no issues with dfu-util on Windows.
If you're willing to help, once again, I strongly suggest checking that pull-up resistor in the first place.
There's always a possibility that we messed up something while adding STM32duino support for PlatformIO. So I would start by testing dfu-util upload with Arduino IDE and [Maple](https://github.com/rogerclarkmelbourne/Arduino_STM32) core first. You would need to install STM32duino bootloader.

jcw, 2018-04-04:
As I said, there's a pull-up, but it's gated with a MOSFET - PC12 has to be set to "0" to enable it, just as on a Maple board, as I understand it.

The behaviour on the Arduino IDE is the same:

* when a sketch with USB console is running, DTR can bring up DFU, and after upload, I have to manually reset to start the sketch
* with "Maple (Rev 3)" as type, the console is present, with "Generic F103R" it's not

My conclusion at this point is that the the DFU code and the USB console code do no always match up, and that's probably what causes most of the issues reported here and on the STM32duino forums. I don't think MacOS differs from Windows in this particular case. When there is no working USB console in your sketch (e.g. the Generic F103 targets), then DTR won't get you to DFU. And if the DFU is not the proper build for the board, then it can't properly launch and cause re-enumeration to switch to a serial USB console.

@ubis - Yes, I can add a 1.5 kΩ pull-up to the Olimexino and bypass the MOSFET. But that's not my point. I would much rather like to see the DFU + serial USB drivers paired up so these issues don't come back on the next board anyone tries, existing or future ones.

@rogerclarkmelbourne - Just wanted to flag this discussion with you, as you know I've traveled this road before - is it me, or is it really too hard to simplify the world of STM32 / F103 / DFU / boards?

ubis, 2018-04-04:
> with "Maple (Rev 3)" as type, the console is present, with "Generic F103R" it's not

Interesting, there might be additional build flags used. I'll look into this once I'll finish adding all supported boards of F1 arch. 

> I can add a 1.5 kΩ pull-up to the Olimexino and bypass the MOSFET

Nah, there's no need for that if R5 on the actual board is 1k5. Also there's no "software" solution for incorrect pull-up resistor value, that's what I meant before.

So yeah, it's not PlatformIO related issue, same as #37