platformio/platform-ststm32 #105: can't compile for board AZ3166 missing mbed_config.h

fmuntean, 2018-04-11:
Just trying to compile an example project for the AZ3166 gives me error: 
Tried both v 4.1 and the git 

PLATFORM: ST STM32 > Microsoft Azure IoT Development Kit (MXChip AZ3166)
SYSTEM: STM32F412ZGT6 100MHz 256KB RAM (1MB Flash)
DEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 9 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WiFi> v1.2.6
|-- <Sensor> v0.0.1
Compiling .pioenvs\dev_mxchip_az3166\src\main.cpp.o
Compiling .pioenvs\dev_mxchip_az3166\lib722\WiFi\AZ3166WiFi.cpp.o
Compiling .pioenvs\dev_mxchip_az3166\lib722\WiFi\AZ3166WiFiClient.cpp.o
Compiling .pioenvs\dev_mxchip_az3166\lib722\WiFi\AZ3166WiFiServer.cpp.o
Compiling .pioenvs\dev_mxchip_az3166\lib722\WiFi\AZ3166WiFiUdp.cpp.o
Compiling .pioenvs\dev_mxchip_az3166\libc74\Sensors\HTS221Sensor.cpp.o
Compiling .pioenvs\dev_mxchip_az3166\libc74\Sensors\IrDASensor.cpp.o
Compiling .pioenvs\dev_mxchip_az3166\libc74\Sensors\LIS2MDLSensor.cpp.o
cc1plus.exe: fatal error: mbed_config.h: No such file or directory

*********************************************************************
* Looking for mbed_config.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:mbed_config.h"
* Web  > https://platformio.org/lib/search?query=header:mbed_config.h
*
*********************************************************************

compilation terminated.
ccc1plus.exe: fatal error: mbed_config.h: No such file or directory

*********************************************************************
* Looking for mbed_config.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:mbed_config.h"
* Web  > https://platformio.org/lib/search?query=header:mbed_config.h
*
*********************************************************************

*** [.pioenvs\dev_mxchip_az3166\src\main.cpp.o] Error 1
compilation terminated.
cc1plus.exe: fatal error: mbed_config.h: No such file or directory

*********************************************************************
* Looking for mbed_config.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:mbed_config.h"
* Web  > https://platformio.org/lib/search?query=header:mbed_config.h
*
*********************************************************************

ccompilation terminated.
cc1plus.exe: fatal error: mbed_config.h: No such file or directory

*********************************************************************
* Looking for mbed_config.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:mbed_config.h"
* Web  > https://platformio.org/lib/search?query=header:mbed_config.h
*
*********************************************************************

compilation terminated.
cc1plus.exe: fatal error: mbed_config.h: No such file or directory

*********************************************************************
* Looking for mbed_config.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:mbed_config.h"
* Web  > https://platformio.org/lib/search?query=header:mbed_config.h
*
*********************************************************************

cc1plus.exe: fatal error: mbed_config.h: No such file or directory

*********************************************************************
* Looking for mbed_config.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:mbed_config.h"
* Web  > https://platformio.org/lib/search?query=header:mbed_config.h

*********************************************************************
* Looking for ?[36mmbed_config.h?[0m dependency? Check our library registry!
*
* CLI  > platformio lib search "header:mbed_config.h"
* Web  > ?[34mhttps://platformio.org/lib/search?query=header:mbed_config.h?[0m
*
*********************************************************************

cc1plus.exe: fatal error: mbed_config.h: No such file or directory

*********************************************************************
* Looking for mbed_config.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:mbed_config.h"
* Web  > https://platformio.org/lib/search?query=header:mbed_config.h
*
*********************************************************************

compilation terminated.
compilation terminated.
compilation terminated.
cc1plus.exe: fatal error: mbed_config.h: No such file or directory

*********************************************************************
* Looking for mbed_config.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:mbed_config.h"
* Web  > h
*********************************************************************
* Looking for ?[36mmbed_config.h?[0m dependency? Check our library registry!
*
* CLI  > platformio lib search "header:mbed_config.h"
* Web  > ?[34mhttps://platformio.org/lib/search?query=header:mbed_config.h?[0m
*
*********************************************************************

compilation terminated.
*** [.pioenvs\dev_mxchip_az3166\lib722\WiFi\AZ3166WiFiUdp.cpp.o] Error 1
*** [.pioenvs\dev_mxchip_az3166\libc74\Sensors\LIS2MDLSensor.cpp.o] Error 1
*** [.pioenvs\dev_mxchip_az3166\libc74\Sensors\IrDASensor.cpp.o] Error 1
*** [.pioenvs\dev_mxchip_az3166\lib722\WiFi\AZ3166WiFiClient.cpp.o] Error 1
*** [.pioenvs\dev_mxchip_az3166\libc74\Sensors\HTS221Sensor.cpp.o] Error 1
*** [.pioenvs\dev_mxchip_az3166\lib722\WiFi\AZ3166WiFi.cpp.o] Error 1
*** [.pioenvs\dev_mxchip_az3166\lib722\WiFi\AZ3166WiFiServer.cpp.o] Error 1

fmuntean, 2018-04-11:
This is the platformio.ini file content:
[env:mxchip_az3166]
platform = ststm32
framework = arduino
board = mxchip_az3166
monitor_baud = 115200

[env:dev_mxchip_az3166]
framework = arduino
board = mxchip_az3166
monitor_baud = 115200
platform = https://github.com/platformio/platform-ststm32.git

ivankravets, 2018-06-06:
Should work now

fmuntean, 2018-06-08:
still does not work after upgrading the platform:
\packages\toolchain-gccarmnoneeabi@1.60301.0\bin\../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/include/c++/6.3.1/arm-none-eabi/thumb/v7e-m/fpv4-sp/softfp
c:\users\fmuntean\.platformio\packages\toolchain-gccarmnoneeabi@1.60301.0\bin\../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/include/c++/6.3.1/backward
c:\users\fmuntean\.platformio\packages\toolchain-gccarmnoneeabi@1.60301.0\bin\../lib/gcc/arm-none-eabi/6.3.1/include
c:\users\fmuntean\.platformio\packages\toolchain-gccarmnoneeabi@1.60301.0\bin\../lib/gcc/arm-none-eabi/6.3.1/include-fixed
c:\users\fmuntean\.platformio\packages\toolchain-gccarmnoneeabi@1.60301.0\bin\../lib/gcc/arm-none-eabi/6.3.1/../../../../arm-none-eabi/include
End of search list.
GNU C++11 (GNU Tools for ARM Embedded Processors 6-2017-q1-update) version 6.3.1 20170215 (release) [ARM/embedded-6-branch revision 245512] (arm-none-eabi)
compiled by GNU C version 5.3.1 20160211, GMP version 6.1.0, MPFR version 3.1.4, MPC version 1.0.3, isl version 0.15
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
Compiler executable checksum: c9153aadcd88080d1765579db058a666
cc1plus.exe: fatal error: mbed_config.h: No such file or directory

*********************************************************************
* Looking for mbed_config.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:mbed_config.h"
* Web  > https://platformio.org/lib/search?query=header:mbed_config.h
*
*********************************************************************

compilation terminated.

fmuntean, 2018-06-08:
I tried with STSTM32 v4.2.1 and the git 49bc388

ivankravets, 2018-06-08:
Could you provide a simple project to reproduce this issue?

fmuntean, 2018-06-08:
I used the sample example for blinking. just added the following on the platformio.ini
[env:mxchip_az3166]
platform = ststm32
board = mxchip_az3166
framework = arduino

[arduino-blink.zip](https://github.com/platformio/platform-ststm32/files/2085497/arduino-blink.zip)

ivankravets, 2018-06-08:
Thanks! 

@valeros please re-test.

fmuntean, 2018-06-11:
I have uninstall everything then reinstall the Visual Studio Code and the STSTM32 platform and was able to compile. Thank You.

closed 2018-06-11 by fmuntean