platformio/platform-ststm32 #106: Time functions are broken (mbed)

OscarGarciaF, 2018-04-11:
The Time functions does not work on bluepill_f103c8, seems to always return -1. Functions are tested with the example code on their website https://os.mbed.com/docs/latest/reference/time.html

Version: Home 0.9.3    Core 3.5.3a8
IDE: Atom
OS: Windows10

Main:
```C++
#include "mbed.h"

Serial pc(B6,B7,57600);//TX, RX,BAUD RATE

int main()
{
    pc.printf("START\n");
    set_time(1256729737);  // Set RTC time to Wed, 28 Oct 2009 11:35:37
        while (true)
        {
            time_t seconds = time(NULL);

            pc.printf("Time as seconds since January 1, 1970 = %d\n", seconds);

            pc.printf("Time as a basic string = %s", ctime(&seconds));

            char buffer[32];
            strftime(buffer, 32, "%I:%M %p\n", localtime(&seconds));
            pc.printf("Time as a custom formatted string = %s", buffer);
            pc.printf("\n");
            wait(1);
        }
}
```
platformio.ini:
```C++
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = mbed
```

OscarGarciaF, 2018-04-14:
Apparently solved here:
https://community.platformio.org/t/time-functions-are-broken-on-bluepill-f103c8-mbed/3952

please check

ivankravets, 2018-04-19:
Thank yo uso much that pointed to the fix!

closed 2018-04-19 by ivankravets