platformio/platform-ststm32 #110: filesystem/mbed_filesystem.h: No such file or directory

OscarGarciaF, 2018-05-16:
I'm having this exact problem: https://github.com/platformio/builder-framework-mbed/issues/7
Posting my .ini

```
[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = mbed
build_flags = -DDEVICE_RTC=0 -D PIO_FRAMEWORK_MBED_RTOS_PRESENT -D PIO_FRAMEWORK_MBED_EVENTS_PRESENT -D PIO_FRAMEWORK_MBED_FILESYSTEM_PRESENT
upload_protocol = stlink
```

Error message is: 

> C:\.platformio\packages\framework-mbed/mbed.h:50:10: fatal error: filesystem/mbed_filesystem.h: No such file or directory
#include "filesystem/mbed_filesystem.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

valeros, 2018-05-16:
Hi @OscarGarciaF! Could you provide you main file? What header file from `filesystem` library do you use?

OscarGarciaF, 2018-05-17:
Hello, the main file is empty:
```
#include <mbed.h>

int main()
{

  for(;;)
  {

  }
}
```
It seems like only having the flags present makes it fail, This is the error message

```
pio run[05/17/18 01:05:06] Processing bluepill_f103c8 (platform: ststm32; board: bluepill_f103c8; frame
work: mbed)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > BluePill F103C8
SYSTEM: STM32F103C8T6 72MHz 20KB RAM (64KB Flash)
DEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 42 compatible libraries
Scanning dependencies...

Dependency Graph
|-- <mbed-rtos>
C:\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py:1303: UnicodeWarning: Unicode equal comparison failed to convert both arguments to Unicode - interpreting them as being
unequal
val = [x for x in val if x not in dk]
Compiling .pioenvs\bluepill_f103c8\FrameworkMbedCore\platform\ATCmdParser.o
Compiling .pioenvs\bluepill_f103c8\FrameworkMbedCore\platform\FileSystemHandle.o
Compiling .pioenvs\bluepill_f103c8\FrameworkMbedCore\platform\mbed_wait_api_rtos.o
In file included from C:\.platformio\packages\framework-mbed\platform\FileSystemHandle.cpp:17:0:
C:\.platformio\packages\framework-mbed/mbed.h:50:10: fatal error: filesystem/mbed_filesystem.h: No such file or directory
#include "filesystem/mbed_filesystem.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from C:\.platformio\packages\framework-mbed\platform\ATCmdParser.h:23:0,
from C:\.platformio\packages\framework-mbed\platform\ATCmdParser.cpp:21:

C:\.platformio\packages\framework-mbed/mbed.h:50:10: fatal error: filesystem/mbed_filesystem.h: No such file or directory
#include "filesystem/mbed_filesystem.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from C:\.platformio\packages\framework-mbed\rtos\RtosTimer.cpp:26:0:
C:\.platformio\packages\framework-mbed/mbed.h:50:10: fatal error: filesystem/mbed_filesystem.h: No such file or directory
#include "filesystem/mbed_filesystem.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
Compiling .pioenvs\bluepill_f103c8\lib1ac\rtos\Thread.o
Compiling .pioenvs\bluepill_f103c8\src\main.o
*** [.pioenvs\bluepill_f103c8\FrameworkMbedCore\platform\FileSystemHandle.o] Error 1
*** [.pioenvs\bluepill_f103c8\FrameworkMbedCore\platform\ATCmdParser.o] Error 1
*** [.pioenvs\bluepill_f103c8\FrameworkMbedCore\platform\mbed_wait_api_rtos.o] Error 1
*** [.pioenvs\bluepill_f103c8\lib1ac\rtos\Kernel.o] Error 1
*** [.pioenvs\bluepill_f103c8\lib1ac\rtos\RtosTimer.o] Error 1
In file included from C:\.platformio\packages\framework-mbed\rtos\Thread.cpp:24:0:
C:\.platformio\packages\framework-mbed/mbed.h:50:10: fatal error: filesystem/mbed_filesystem.h: No such file or directory

#include "filesystem/mbed_filesystem.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from src\main.cpp:1:0:
C:\.platformio\packages\framework-mbed/mbed.h:50:10: fatal error: filesystem/mbed_filesystem.h: No such file or directory
#include "filesystem/mbed_filesystem.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

*** [.pioenvs\bluepill_f103c8\lib1ac\rtos\Thread.o] Error 1
*** [.pioenvs\bluepill_f103c8\src\main.o] Error 1
================================================================================== [ERROR] Took 8.12 seconds ==================================================================================
```

OscarGarciaF, 2018-05-17:
If i add it then it does compile:
```
#include <mbed.h>
#include "filesystem/mbed_filesystem.h"
int main()
{
  for(;;)
  {

  }
}
```
However if i do this:
```
#include <mbed.h>
#include "filesystem/mbed_filesystem.h"
#include <Debug.h>
int main()
{
  for(;;)
  {

  }
}
```
then it shows the following error
```
In file included from lib\Debug\Debug.cpp:1:0:
C:\.platformio\packages\framework-mbed/mbed.h:50:10: fatal error: filesystem/mbed_filesystem.h: No such file or directory
#include "filesystem/mbed_filesystem.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\bluepill_f103c8\lib81b\Debug\Debug.o] Error 1
```
where Debug.cpp:
```
#include <mbed.h>
#include "Debug.h"

Serial debugSerial(PA_2, PA_3, 57600);
```
And Debug.h:
```
#ifndef DEBUGSERIAL_H
#define DEBUGSERIAL_H

#include <mbed.h>

extern Serial debugSerial;

#endif // DEBUGSERIAL_H
```
adding #include "filesystem/mbed_filesystem.h" to Debug.cpp solves the problem but i dont think this is a correct behaviour of the compiler (nor i want to be adding headers to every one of my .cpps)

ivankravets, 2018-05-31:
Relatedissue https://github.com/platformio/builder-framework-mbed/issues/7

@valeros do you have any ideas?

ivankravets, 2018-06-04:
Could you re-test with the latest upstream version? 

- http://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions

closed 2018-06-04 by ivankravets