platformio/platform-ststm32 #112: Upload_protocol = DFU cannot find the firmware file if project path contains spaces

me21, 2018-05-29:
In the output during upload a truncated project path can be seen (up to the first space).
Removing the spaces from directory name fixed this for me.

ivankravets, 2018-05-30:
@ubis why do we need [here](https://github.com/platformio/platform-ststm32/blob/develop/builder/main.py#L199) `$PROJECT_DIR/`? It should work as `UPLOADCMD="$UPLOADER $UPLOADERFLAGS $SOURCES"`.

Could you re-test it? Thanks!

ubis, 2018-05-30:
Seems like `maple_upload` needs the full path, because I get:

```
CURRENT: upload_protocol = dfu
MethodWrapper(["upload"], [".pioenvs\bluepill_f103c8\firmware.bin"])
Auto-detected: COM5
maple_upload COM5 2 1EAF:0003 .pioenvs\bluepill_f103c8\firmware.bin
maple_loader v0.1
Resetting to bootloader via DTR pulse
Searching for DFU device [1EAF:0003]...
Found it!
.pioenvs\bluepill_f103c8\firmware.bin: No such file or directory
```

ivankravets, 2018-05-30:
Could you try this?
```
UPLOADCMD='$UPLOADER $UPLOADERFLAGS "$PROJECT_DIR/$SOURCES"'
```

ubis, 2018-05-31:
That seems to work. Although I couldn't reproduce problem with having directory with spaces. Still, by using this, it should work even with spaces.

commit 671ff510 (2018-06-04): Escape path to the firmware when upload_protocol = dfu // Resolve #112
closed 2018-06-04 by ivankravets