platformio/platform-ststm32 #113: Installed library include paths being appended to framework compile

MrSurly, 2018-05-29:
### Configuration

**Operating system**: Ubuntu 18.04

**PlatformIO Version** (`platformio --version`): PlatformIO, version 3.5.2

### Description of problem

Adding the `SX1276Lib` (2757) causes a compile error in the MBED framework.

#### Steps to Reproduce

1. `pio init --board disco_l072cz_lrwan1`
2. `echo >>platformio.ini 'lib_deps=2757'`
3. `echo >src/main.cpp 'int main() {}'`
4. `pio run`

### Actual Results

```
[Tue May 29 12:17:31 2018] Processing disco_l072cz_lrwan1 (platform: ststm32; board: disco_l072cz_lrwan1; framework: mbed)
---------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > ST DISCO-L072CZ-LRWAN1
SYSTEM: STM32L072CZ 32MHz 20KB RAM (192KB Flash)
DEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(light)
Collected 16 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SX1276Lib>
Compiling .pioenvs/disco_l072cz_lrwan1/FrameworkMbedCore/drivers/FlashIAP.o
Compiling .pioenvs/disco_l072cz_lrwan1/FrameworkMbedCore/drivers/InterruptManager.o
Compiling .pioenvs/disco_l072cz_lrwan1/FrameworkMbedCore/drivers/MbedCRC.o
Compiling .pioenvs/disco_l072cz_lrwan1/FrameworkMbedCore/drivers/RawSerial.o
In file included from /home/epoulsen/.platformio/packages/framework-mbed/platform/NonCopyable.h:21:0,
from /home/epoulsen/.platformio/packages/framework-mbed/platform/PlatformMutex.h:26,
from /home/epoulsen/.platformio/packages/framework-mbed/drivers/FlashIAP.h:29,
from /home/epoulsen/.platformio/packages/framework-mbed/drivers/FlashIAP.cpp:26:
/home/epoulsen/.platformio/packages/framework-mbed/platform/mbed_debug.h: In function 'void debug(const char*, ...)':
/home/epoulsen/.platformio/packages/framework-mbed/platform/mbed_debug.h:40:20: error: redefinition of 'void debug(const char*, ...)'
static inline void debug(const char *format, ...) {
^~~~~
In file included from /home/epoulsen/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/bits/stl_algobase.h:69:0,
from /home/epoulsen/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/7.2.1/algorithm:61,
from /home/epoulsen/.platformio/packages/framework-mbed/drivers/FlashIAP.cpp:25:
.piolibdeps/SX1276Lib_ID2757/debug/debug.h:56:20: note: 'void debug(const char*, ...)' previously defined here
static inline void debug(const char *format, ...) {}
^~~~~
*** [.pioenvs/disco_l072cz_lrwan1/FrameworkMbedCore/drivers/FlashIAP.o] Error 1
======================================= [ERROR] Took 0.79 seconds =======================================
```

### Expected Results

Error-free compilation.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:disco_l072cz_lrwan1]
platform = ststm32
board = disco_l072cz_lrwan1
framework = mbed
lib_deps=2757
```

**Source file to reproduce issue:**
```cpp
int main() {}
```

### Additional info

Also posted this issue [here](https://community.platformio.org/t/sx1276lib-not-compatible-with-mbed-framework/4170). 

Problem seems to be that PIO will append _library_ include directories when compiling framework source code, which doesn't seem useful -- the framework shouldn't rely on library includes.

Abbreviated command line:

```bash
arm-none-eabi-g++ \
-o .pioenvs/disco_l072cz_lrwan1/FrameworkMbedCore/drivers/Serial.o \
<redacted> \
-I/home/epoulsen/.platformio/packages/framework-mbed \
<other includes> \
-I/home/epoulsen/.platformio/packages/framework-mbed/platformio/variants/DISCO_L072CZ_LRWAN1 \
-Isrc \
-I.piolibdeps/SX1276Lib_ID2757 \
/home/epoulsen/.platformio/packages/framework-mbed/drivers/Serial.cpp
```

Specifically, this: `-I.piolibdeps/SX1276Lib_ID2757`

Full command line:
```bash
arm-none-eabi-g++ -o .pioenvs/disco_l072cz_lrwan1/FrameworkMbedCore/drivers/Serial.o -c -std=gnu++98 -fno-rtti -Wvla -mcpu=cortex-m0plus -mthumb -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -DNDEBUG -g1 -include mbed_config.h -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DMBED_BUILD_TIMESTAMP=1524075067.19 -DDEVICE_ANALOGOUT=1 -DTARGET_STM32L072CZ -DTOOLCHAIN_GCC -DTARGET_STM32L0 -DTARGET_CORTEX_M -DTARGET_M0P -DDEVICE_TRNG=1 -DTARGET_UVISOR_UNSUPPORTED -DTARGET_STM32L072xx -DDEVICE_SPI_ASYNCH=1 -DDEVICE_PWMOUT=1 -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M0PLUS -DDEVICE_STDIO_MESSAGES=1 -DTARGET_FF_MORPHO -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DTARGET_STM -DDEVICE_SERIAL_FC=1 -DTARGET_DISCO_L072CZ_LRWAN1 -DARM_MATH_CM0PLUS -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 -DDEVICE_I2C_ASYNCH=1 -DTARGET_STM32L072xZ -DTARGET_LIKE_CORTEX_M0 -DSTM32L072xx -DPLATFORMIO=30502 -I/home/epoulsen/.platformio/packages/framework-mbed -I/home/epoulsen/.platformio/packages/framework-mbed/cmsis -I/home/epoulsen/.platformio/packages/framework-mbed/cmsis/TARGET_CORTEX_M -I/home/epoulsen/.platformio/packages/framework-mbed/drivers -I/home/epoulsen/.platformio/packages/framework-mbed/hal -I/home/epoulsen/.platformio/packages/framework-mbed/hal/storage_abstraction -I/home/epoulsen/.platformio/packages/framework-mbed/platform -I/home/epoulsen/.platformio/packages/framework-mbed/targets -I/home/epoulsen/.platformio/packages/framework-mbed/targets/TARGET_STM -I/home/epoulsen/.platformio/packages/framework-mbed/targets/TARGET_STM/TARGET_STM32L0 -I/home/epoulsen/.platformio/packages/framework-mbed/targets/TARGET_STM/TARGET_STM32L0/device -I/home/epoulsen/.platformio/packages/framework-mbed/targets/TARGET_STM/TARGET_STM32L0/TARGET_STM32L072xZ -I/home/epoulsen/.platformio/packages/framework-mbed/targets/TARGET_STM/TARGET_STM32L0/TARGET_STM32L072xZ/device -I/home/epoulsen/.platformio/packages/framework-mbed/targets/TARGET_STM/TARGET_STM32L0/TARGET_STM32L072xZ/device/TOOLCHAIN_GCC_ARM -I/home/epoulsen/.platformio/packages/framework-mbed/targets/TARGET_STM/TARGET_STM32L0/TARGET_STM32L072xZ/TARGET_DISCO_L072CZ_LRWAN1 -I/home/epoulsen/.platformio/packages/framework-mbed -I/home/epoulsen/.platformio/packages/framework-mbed/platformio/variants/DISCO_L072CZ_LRWAN1 -Isrc -I.piolibdeps/SX1276Lib_ID2757 /home/epoulsen/.platformio/packages/framework-mbed/drivers/Serial.cpp
```

valeros, 2018-06-06:
Hi @MrSurly! Actually, appending the library path is not the issue.
Here are a few points that prevent this library from compiling correctly:
1. The `SX1276Lib` library has the next path in its structure `debug/debug.h`. Unfortunately, this path is also used in one of the internal files from the GCC compiler (`stl_iterator_base_funcs.h:65` to be precise) and thus build system decided to pick this file instead of default one.
2. Currently, PlatformIO build system has basic support for mbed libraries, especially for those that have an unusual source files structure, so some of them require additional modifications to be used with PlatformIO. For example, for this library, you need to add extra include paths to `library.json`

So, I'd recommend the next workaround:
1. Rename file `debug/debug.h` to `debug/_debug.h` in the library folder.
2. Add extra include paths to the `library.json` file
```
{
    .... 
    "repository": {
        "type": "hg",
        "url": "https://os.mbed.com/teams/Semtech/code/SX1276Lib/"
    },
    "build": {
        "flags": "-Iradio -Ienums -Iregisters -Iregisters -Isx1276 -Itypedefs"
    }
}
```

MrSurly, 2018-06-07:
Hello @valeros.

> Actually, appending the library path is not the issue.

I'd argue that there's no valid use case for appending library include paths to framework builds.  Frameworks are never (or shouldn't be) dependent on libraries, thus these paths should never be added when building framework files.

I do understand that the library itself has issues.

> So, I'd recommend the next workaround:

One of the major advantages of using PlatformIO in the first place is that it allows workspaces to be easily shared.  Libraries installed via `lib_deps` in `platformio.ini` shouldn't need after-the-fact workarounds. especially considering `.piolibdeps` is hidden, and in `.gitignore`.

I realize libraries are (for the most part) published by 3rd parties -- is there no vetting process? This particular library is clearly not at all compatible with PlatformIO, since it only supports 1 framework, and won't compile on that framework at all.

ivankravets, 2018-06-07:
> I'd argue that there's no valid use case for appending library include paths to framework builds. 

This is a bug, we will fix it in PIO Core soon.

> This particular library is clearly not at all compatible with PlatformIO

We will improve support for old-style mbed libraries. 

I'll inform you here about further updates.

MrSurly, 2018-06-07:
> We will improve support for old-style mbed libraries.

Can I help, or is this something the lib maintainer has to do?

ivankravets, 2018-06-07:
> Can I help, or is this something the lib maintainer has to do?

Thanks! I opened 2 issues. You are added to CC.

ivankravets, 2018-06-11:
OK, we fix that issue partially on our side. Another part of the issue will be fixed soon. Nevertheless, you don't need to wait for the last fix.

1. Please open PIO IDE Terminal in IDE and type `pio upgrade --dev`
2. Use the next `platformio.ini`

```ini
[env:disco_l072cz_lrwan1]
platform = ststm32
board = disco_l072cz_lrwan1
framework = mbed
lib_deps = https://os.mbed.com/teams/Semtech/code/SX1276Lib/
```

closed 2018-06-11 by ivankravets