platformio/platform-ststm32 #117: Multiple SPI on the bluepill_f103c8 with mbed

OscarGarciaF, 2018-06-11:
I'm trying to use an ethernet [library](https://os.mbed.com/users/hudakz/code/UIPEthernet/file/4acb22344932/utility/) that is using spi to communicate over the internet, everything was working fine until i added a datalogger with a SD card (that is also using a SPI) and my ethernet connection became inestable.
Further testing revealed that the Mutex, CS pins and SPI.locks were not the problem as everything was being called appropriately.

Afterwards i separated the two SPIs into the SPI1 and SPI2 channels to avoid synchronizations problems and the problem still persisted, just the existance of another SPI instance is causing stability problems.

Does anyone have any ideas on how to resolve this?

ivankravets, 2019-02-21:
Does it work with the upstream version? See docs https://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions

OscarGarciaF, 2019-02-21:
I stopped working on this project months ago and no longer have access to it.
At the end i discovered that the build specifically did not worked on my computer and it worked on a coworker computer without issue

I have windows 10 64 bits

closed 2019-02-21 by ivankravets