platformio/platform-ststm32 #118: Uploading in dfu mode for STMF407VET6

masoudr, 2018-06-15:
Hi,
I recently got a brand new STM32F407VET6 board [link here](http://wiki.stm32duino.com/index.php?title=STM32F407). I tried to choose `stm32F704VET6 generic` for the projects'board. I want to program it through dfu mode and set the jumpers properly and board came up as an STM DFU device. I also can program the board with ST app with no problem but when I want to upload with platformio I get this error:

> Couldn't find the DFU device: [1EAF:0003]

And here is my config file:
```
[env:genericSTM32F407VET6]
platform = ststm32
board = genericSTM32F407VET6
framework = arduino
upload_protocol = dfu
```
Thanks.

ubis, 2018-06-15:
Hi, which operating system you're using?

> I also can program the board with ST app
You mean st-link, or it's some sort of st-dfu uploader?

Also, have you tried with Arduino IDE and [maple core](https://github.com/rogerclarkmelbourne/Arduino_STM32)?

masoudr, 2018-06-15:
I'm using Windows 10.
The DefuseDemo app from ST which works on dfu mode can program the chip fine and if I compile my code and convert it to *.dfu I can program it without any problem.
I tried with Arduino IDE and getting the same problem but In the IDE there is no option to choose the DFU mode, the only option is using `STMduino Bootloader` which didn't work for me (Failed on checking for a serial port which STM don't have it on DFU mode), I also tried to use a USB to Serial but with no luck. I think there is a lack of support for STM32F407 devices. I also found [this](http://www.stm32duino.com/viewtopic.php?f=39&t=2663) and it upload the code which seems works fine but I don't know how to make it compatible with Platformio.
The only method that works fine for me was another core `STM32GENERIC` link [here](https://github.com/danieleff/STM32GENERIC). But I don't know how to bring it to Platformio.
I think there are three options to program an STM32,
1. First using ST-Link
2. Using throw DFU mode with USB connection
3. Using a USB to Serial Cable with UART1 pins.

ubis, 2018-06-15:
Can you provide the successful upload logs with `STM32GENERIC` core? The scripts used in this and `Maple` core are the same, yet I can't tell why it's not working with `Maple` core.

masoudr, 2018-06-15:
I connected my USB to Serial cable to UART1 and set the upload method to `Serial UART1`. Here is complete log for a successful upload:

> "C:\Users\Masoud\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-ar" rcs  "C:\Users\Masoud\AppData\Local\Temp\arduino_build_331974\core\core.a" "C:\Users\Masoud\AppData\Local\Temp\arduino_build_331974\core\stm32\USBDevice.cpp.o"
> Archiving built core (caching) in: C:\Users\Masoud\AppData\Local\Temp\arduino_cache_661225\core\core_STM32GENERIC_STM32_BLACK_F407XX_upload_method_serialMethod,usb_SerialUSB,serial_Automatic,subboard_BLACK_F407VE_f7d77c593ff2a5d23eea302bdeae6336.a
> Linking everything together...
> "C:\Users\Masoud\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-gcc" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb -Os -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align "-TC:\Users\Masoud\Documents\Arduino\hardware\STM32GENERIC\STM32\variants\BLACK_F407VE/ldscript.ld" "-Wl,-Map,C:\Users\Masoud\AppData\Local\Temp\arduino_build_331974/STM32_Blinky2.ino.map"  -o "C:\Users\Masoud\AppData\Local\Temp\arduino_build_331974/STM32_Blinky2.ino.elf" "-LC:\Users\Masoud\AppData\Local\Temp\arduino_build_331974" -Wl,--start-group "C:\Users\Masoud\AppData\Local\Temp\arduino_build_331974\sketch\STM32_Blinky2.ino.cpp.o" "C:\Users\Masoud\AppData\Local\Temp\arduino_build_331974\core\variant.c.o" "C:\Users\Masoud\AppData\Local\Temp\arduino_build_331974/core\core.a" -lc -Wl,--end-group -lm -lgcc --specs=nano.specs
> "C:\Users\Masoud\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-objcopy" -O binary  "C:\Users\Masoud\AppData\Local\Temp\arduino_build_331974/STM32_Blinky2.ino.elf" "C:\Users\Masoud\AppData\Local\Temp\arduino_build_331974/STM32_Blinky2.ino.bin"
> Sketch uses 13168 bytes (2%) of program storage space. Maximum is 524288 bytes.
> Global variables use 3064 bytes (2%) of dynamic memory, leaving 128008 bytes for local variables. Maximum is 131072 bytes.
> stm32flash 0.4
> 
> http://stm32flash.googlecode.com/
> 
> Using Parser : Raw BINARY
> Interface serial_w32: 230400 8E1
> Version      : 0x31
> Option 1     : 0x00
> Option 2     : 0x00
> Device ID    : 0x0413 (STM32F40/1)
> - RAM        : 128KiB  (8192b reserved by bootloader)
> - Flash      : 1024KiB (sector size: 4x16384)
> - Option RAM : 16b
> - System RAM : 29KiB
> Write to memory
> Erasing memory
> 
> Wrote address 0x08000100 (1.94%) 
> Wrote address 0x08000200 (3.88%) 
> Wrote address 0x08000300 (5.83%) 
> Wrote address 0x08000400 (7.77%) 
> Wrote address 0x08000500 (9.71%) 
> Wrote address 0x08000600 (11.65%) 
> Wrote address 0x08000700 (13.60%) 
> Wrote address 0x08000800 (15.54%) 
> Wrote address 0x08000900 (17.48%) 
> Wrote address 0x08000a00 (19.42%) 
> Wrote address 0x08000b00 (21.37%) 
> Wrote address 0x08000c00 (23.31%) 
> Wrote address 0x08000d00 (25.25%) 
> Wrote address 0x08000e00 (27.19%) 
> Wrote address 0x08000f00 (29.14%) 
> Wrote address 0x08001000 (31.08%) 
> Wrote address 0x08001100 (33.02%) 
> Wrote address 0x08001200 (34.96%) 
> Wrote address 0x08001300 (36.90%) 
> Wrote address 0x08001400 (38.85%) 
> Wrote address 0x08001500 (40.79%) 
> Wrote address 0x08001600 (42.73%) 
> Wrote address 0x08001700 (44.67%) 
> Wrote address 0x08001800 (46.62%) 
> Wrote address 0x08001900 (48.56%) 
> Wrote address 0x08001a00 (50.50%) 
> Wrote address 0x08001b00 (52.44%) 
> Wrote address 0x08001c00 (54.39%) 
> Wrote address 0x08001d00 (56.33%) 
> Wrote address 0x08001e00 (58.27%) 
> Wrote address 0x08001f00 (60.21%) 
> Wrote address 0x08002000 (62.15%) 
> Wrote address 0x08002100 (64.10%) 
> Wrote address 0x08002200 (66.04%) 
> Wrote address 0x08002300 (67.98%) 
> Wrote address 0x08002400 (69.92%) 
> Wrote address 0x08002500 (71.87%) 
> Wrote address 0x08002600 (73.81%) 
> Wrote address 0x08002700 (75.75%) 
> Wrote address 0x08002800 (77.69%) 
> Wrote address 0x08002900 (79.64%) 
> Wrote address 0x08002a00 (81.58%) 
> Wrote address 0x08002b00 (83.52%) 
> Wrote address 0x08002c00 (85.46%) 
> Wrote address 0x08002d00 (87.41%) 
> Wrote address 0x08002e00 (89.35%) 
> Wrote address 0x08002f00 (91.29%) 
> Wrote address 0x08003000 (93.23%) 
> Wrote address 0x08003100 (95.17%) 
> Wrote address 0x08003200 (97.12%) 
> Wrote address 0x08003300 (99.06%) 
> Wrote address 0x0800337c (100.00%) Done.
> 
> Starting execution at address 0x08000000... done.
>

ubis, 2018-06-15:
Please enable verbose logging by going to `File -> Preferences` and on `Show verbose output during:` enable `upload` in Arduino IDE and post the log again, I need to know the exact command for upload action.

ubis, 2018-06-15:
Not really. I was looking for this line `C:\Users\Masoud\Documents\Arduino\hardware\STM32GENERIC/tools/win/serial_upload.bat COM6 {upload.altID} {upload.usbID} C:\Users\Masoud\AppData\Local\Temp\arduino_build_768183/STM32_Blinky2.ino.bin`

Hmm that looks like `serial` upload protocol rather than `dfu`
Could you try to use `upload_protocol = serial`?

masoudr, 2018-06-15:
Thanks. It worked with the serial method.
Here is the configuration:
```
[env:genericSTM32F407VET6]
platform = ststm32
board = genericSTM32F407VET6
framework = arduino
upload_protocol = serial
upload_port = COM6
```
It solved my problem but when I choose the dfu mode in `upload_protocol` I see this message which is not valid because the dfu mode has no serial port:

> Error: Please specify `upload_port` for environment or use global `--upload-port` option.
> For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
> *** [upload] Explicit exit, status 1
>  [ERROR] Took 7.04 seconds
>  
>

masoudr, 2018-06-16:
I tried that method but It didn't work for me. The output print the successful upload but the board never run the code, I think there is a problem in upload method. Anyway, I think the best method here to use a USB to Serial cable which works fine and the `STM32GENERIC` has better examples and compatibility for F4 series compared to the original version. [THIS](http://stm32duino.com/viewtopic.php?t=2066) reference demonstrates how to bring it to Platformio and worked fine for serial upload.

closed 2018-08-02 by masoudr

jcw, 2018-11-10:
I'd be very interested to help figure out how to use DFU upload w/ the STM32F407's built-in boot loader.

I have an F407 (board `genericSTM32F407VET6`), and can place it in DFU mode by powering it up with BOOT0 set to 1. It shows up as "STM32 BOOTLOADER" in MacOS. When using a self-installed "dfu-util", it shows up as:

```
$ dfu-util -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [0483:df11] ver=2200, devnum=2, cfg=1, intf=0, path="20-6", alt=3, name="@Device Feature/0xFFFF0000/01*004 e", serial="3754315A3436"
Found DFU: [0483:df11] ver=2200, devnum=2, cfg=1, intf=0, path="20-6", alt=2, name="@OTP Memory /0x1FFF7800/01*512 e,01*016 e", serial="3754315A3436"
Found DFU: [0483:df11] ver=2200, devnum=2, cfg=1, intf=0, path="20-6", alt=1, name="@Option Bytes  /0x1FFFC000/01*016 e", serial="3754315A3436"
Found DFU: [0483:df11] ver=2200, devnum=2, cfg=1, intf=0, path="20-6", alt=0, name="@Internal Flash  /0x08000000/04*016Kg,01*064Kg,07*128Kg", serial="3754315A3436"
```

With a little help, PIO will in fact find the device:

```
$ pio run -t upload --upload-port 0483:df11
Processing genericSTM32F407VET6 (platform: ststm32; board: genericSTM32F407VET6; framework: arduino)
-------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VET6.html
PLATFORM: ST STM32 > STM32F407VE (192k RAM. 512k Flash)
SYSTEM: STM32F407VET6 168MHz 128KB RAM (502.23KB Flash)
DEBUG: CURRENT(stlink) EXTERNAL(stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 6 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pioenvs/genericSTM32F407VET6/firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   1.1% (used 1440 bytes from 131072 bytes)
PROGRAM: [          ]   1.2% (used 6132 bytes from 514288 bytes)
Configuring upload protocol...
AVAILABLE: dfu, stlink
CURRENT: upload_protocol = dfu
Looking for upload port...
Use manually specified: 0483:df11
Uploading .pioenvs/genericSTM32F407VET6/firmware.bin
Failed to open serial device.
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
dfu-util: Invalid DFU suffix signature
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!

Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #2 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
dfu-util: Non-valid multiplier ' ', assuming bytes
dfu-util: Non-valid multiplier ' ', assuming bytes
Device returned transfer size 2048
DfuSe interface name: "OTP Memory "
dfu-util: Only DfuSe file version 1.1a is supported
dfu-util: (for raw binary download, use the --dfuse-address option)
*** [upload] Error 74
============================= [ERROR] Took 0.81 seconds =============================
```

My gut feeling is that this is only a **tiny** step away from being able to upload via PIO to it, with nothing more than a USB cable.

jcw, 2018-11-10:
To follow up - this manual upload with dfu-util works:

```
$ dfu-util -d 0483:df11 -a 0 -s 0x08000000 -D .pioenvs/genericSTM32F407VET6/firmware.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 6132
Download	[=========================] 100%         6132 bytes
Download done.
File downloaded successfully
$
```

(needs a reset with BOOT0 set back to 0 to start running)

mklengel, 2020-11-30:
Same problem for my board:

Hello,

I'm trying to upload a Arduino program with to the following board:

BLACK_F407VE.
Can be found at GitHub, user is "mcauser" (I'm not allowed to put links in this message as I'm a new user). The specs can be found in the link above, the board has 512 KB Flash, 192 + 4 KB RAM.
On the backside its labeled with "STM32_F4VE, V2.0 1509"

dfu-util -l reports this:
Found DFU: [0483:df11] ver=2200, devnum=4, cfg=1, intf=0, path="253-5.2", alt=3, name="@Device Feature/0xFFFF0000/01*004 e", serial="207A389B5848"
Found DFU: [0483:df11] ver=2200, devnum=4, cfg=1, intf=0, path="253-5.2", alt=2, name="@OTP Memory /0x1FFF7800/01*512 e,01*016 e", serial="207A389B5848"
Found DFU: [0483:df11] ver=2200, devnum=4, cfg=1, intf=0, path="253-5.2", alt=1, name="@Option Bytes  /0x1FFFC000/01*016 e", serial="207A389B5848"
Found DFU: [0483:df11] ver=2200, devnum=4, cfg=1, intf=0, path="253-5.2", alt=0, name="@Internal Flash  /0x08000000/04*016Kg,01*064Kg,07*128Kg", serial="207A389B5848"

Here is my platform.ini file:

; PlatformIO Project Configuration File
[env:black_f407ve]
platform = ststm32
board = black_f407ve
framework = arduino
upload_protocol = dfu

During the upload with dfu-util I see the wrong memory size (128 kB RAM instead of 196 kB). This values are hard coded in boards/black_f407ve.json and for my board wrong.

Processing black_f407ve (platform: ststm32; board: black_f407ve; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/black_f407ve.html
PLATFORM: ST STM32 (9.0.0) > Black STM32F407VE
HARDWARE: STM32F407VET6 168MHz, 128KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 4.10900.200819 (1.9.0) 
 - framework-cmsis 2.50501.200527 (5.5.1) 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1000.200630 (10.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/black_f407ve/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.7% (used 940 bytes from 131072 bytes)
Flash: [          ]   2.3% (used 12092 bytes from 524288 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, dfu, jlink, serial, stlink
CURRENT: upload_protocol = dfu
Uploading .pio/build/black_f407ve/firmware.bin
dfu-util 0.9
dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
[... 27 lines cut ...]

Terminal will be reused by tasks, press any key to close it.

Using instead genericSTM32F407VET6 shows the correct RAM but dfu-util tries to upload to the wrong vendor/product id ("-d 0x1EAF:0x0003,0x0483:0x3748" instead of "-d 0483:df11":

Processing genericSTM32F407VET6 (platform: ststm32; board: genericSTM32F407VET6; framework: arduino; upload_protocol: dfu)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VET6.html
PLATFORM: ST STM32 (9.0.0) > STM32F407VE (192k RAM. 512k Flash)
HARDWARE: STM32F407VET6 168MHz, 128KB RAM, 502.23KB Flash
.
.
.
"/Users/mklengel/.platformio/packages/tool-dfuutil/bin/dfu-util" -d 0x1EAF:0x0003,0x0483:0x3748 -a 0 -s 0x08000000:leave -D "/Volumes/Macintosh SR0/Users/mklengel/Development/PlatformIO/mkBlinkySTM32F407VE/.pio/build/genericSTM32F407VET6/firmware.bin"
dfu-util 0.9

dfu-util: Invalid DFU suffix signature
Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: No DFU capable USB device available
*** [upload] Error 74

valeros, 2020-11-30:
Hi @mklengel, you can can override the default values directly in your `platformio.ini`:
```ini
[env:black_f407ve]
platform = ststm32
board = black_f407ve
framework = arduino
upload_protocol = dfu
board_upload.maximum_ram_size= 200704
```

mklengel, 2020-12-01:
Thx - this did the trick.