platformio/platform-ststm32 #121: Problem With maple_ret6

Dark-Guan, 2018-06-20:
I use maple_ret6  ,but the program compiled with "board = maple_ret6" can not run with bootloader to 0x08005000, but With "board = maple_ret6" It runs OK.

As well .
/*
 * SDIO Pins
 */
#define BOARD_SDIO_D0 			PC8
#define BOARD_SDIO_D1 			PC9
#define BOARD_SDIO_D2 			PC10
#define BOARD_SDIO_D3 			PC11
#define BOARD_SDIO_CLK 			PC12
#define BOARD_SDIO_CMD 			PD2
should to be add to maple_ret6 's "board.h" to fix some error

Dark-Guan, 2018-06-20:
First 16 bytes form bin file with "board = maple_ret6" It can not be set up
00	00	01	20	A1	01	01	08	A1	04	01	08	A5	04	01	08
First 16 bytes form bin file with "board = maple" It runs OK
00	50	00	20	5D	51	00	08	61	53	00	08	65	53	00	08

Dark-Guan, 2018-06-21:
@ubis @ivankravets It seems that STM32F103RET6 is not work well with STM32duino-bootloader, Neither board=maple_ret6 or board = genericSTM32F103RET6, The bootloader can't jump to correct point to run the APP up. 
I think there may be some different things between RET6 MCU and RBT6 MUC make this wried things

ubis, 2018-06-21:
It seems that it uses correct ldscript, and start address is right. Could you check it with Arduino IDE first?

ivankravets, 2018-12-11:
Please reopen if you still need help

closed 2018-12-11 by ivankravets

Dark-Guan, 2018-12-17:
Thanks all ,I have soled the problem It is the problem with STM32duino maple lib