platformio/platform-ststm32 #123: Debugging error in mbed framework

masoudr, 2018-06-23:
Hi,
I'm using an STM32F407V board with mbed platform. I have ST-Link programmer and I can program the board without any problem. Now I'm trying to use debugging mode but when I press the debug I see this error message:

> Error launching PIO Debugger
> Error: .pioinit:12: Error in sourced command file: Remote communication error. Target disconnected.: No error.

And:

> GNU ARM Eclipse Open on-chip Debugger 0.10 ....
> ...
> Info: The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
> adapter speed: 2000 kHz
> adapter_delay: 100
> none separated
> Info: tcl server disabled
> Info: telnet server disabled
> Info: Unable to match request speed 2000 kHz, using 1800 kHz
> Info: Unable to match request speed 2000 kHz, using 1800 kHz
> Info: clock speed 1800 kHz
> Error: Open faild
> in procedure `init`
> in procedure `ocd_bouncer`
> 

Here is my env config:

```
[env:seeedArchMax]
platform = ststm32
board = seeedArchMax
framework = mbed
;upload_port = COM10
upload_protocol = stlink

[env:debug]
platform = ststm32
upload_protocol = stlink
framework = mbed
board = seeedArchMax
debug_port = localhost:3333
debug_tool = custom
debug_load_cmd = C:/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gdb.exe
debug_server =
  $PLATFORMIO_HOME_DIR/packages/tool-openocd/bin/openocd.exe
  -f $PLATFORMIO_HOME_DIR/packages/tool-openocd/scripts/interface/stlink-v2-1.cfg
  -f
  $PLATFORMIO_HOME_DIR/packages/tool-openocd/scripts/target/stm32f4x.cfg
```

ivankravets, 2018-06-28:
Could you try without these lines?
```
debug_port = localhost:3333
debug_tool = custom
debug_load_cmd = C:/Program Files (x86)/GNU Tools ARM Embedded/7 2017-q4-major/bin/arm-none-eabi-gdb.exe
debug_server =
  $PLATFORMIO_HOME_DIR/packages/tool-openocd/bin/openocd.exe
  -f $PLATFORMIO_HOME_DIR/packages/tool-openocd/scripts/interface/stlink-v2-1.cfg
  -f
  $PLATFORMIO_HOME_DIR/packages/tool-openocd/scripts/target/stm32f4x.cfg
```

It should work automatically.

karanjamutahi, 2019-08-09:
Reopening this because I have the same error.

Platformio.ini looks like so:
```
[env:nucleo_f103rb]
platform = ststm32
board = nucleo_f103rb
framework = mbed
```

Error I got:
```
Error: init mode failed (unable to connect to the target)

.pioinit:12: Error in sourced command file:
```

ivankravets, 2019-08-11:
@karanjamutahi does uploading work? `pio run -t upload`?