platformio/platform-ststm32 #126: firmware can only be uploaded once

mr-freeze, 2018-06-25:
### Configuration

**Operating system**: Windows 10

**PlatformIO Version**:
$ pio --version
PlatformIO, version 3.6.0a3

**Hardware**:

- maple mini clone
- connected via USB
- installed with the [STM32duino-bootloader](https://github.com/rogerclarkmelbourne/STM32duino-bootloader/tree/master/binaries)

### Description of problem

When uploading a file `firmware.bin` to the board only the first flash process is successful. Every next try fails (see error message below). The LED flashes quickly until the MCU is reset. To program another application, the bootloader must first be reinstalled.

#### Steps to Reproduce

1. $ pio run -e maple_mini_b20 -t upload
2. $ pio run -e maple_mini_b20 -t upload

### Actual Results

> CURRENT: upload_protocol = dfu
> Looking for upload port...
> Auto-detected: COM11
> Uploading .pioenvs\maple_mini_b20\firmware.bin
> maple_loader v0.1
> Resetting to bootloader via DTR pulse
> Searching for DFU device [1EAF:0003]...
> dfu-util - (C) 2007-2008 by OpenMoko Inc.
> Couldn't find the DFU device: [1EAF:0003]T
> his program is Free Software and has ABSOLUTELY NO WARRANTY

### Expected Results

> CURRENT: upload_protocol = dfu
> Looking for upload port...
> Auto-detected: COM11
> Uploading .pioenvs\maple_mini_b20\firmware.bin
> maple_loader v0.1
> Resetting to bootloader via DTR pulse
> Searching for DFU device [1EAF:0003]...
> Found it!

> Opening USB Device 0x1eaf:0x0003...
> Found Runtime: [0x1eaf:0x0003] devnum=1, cfg=0, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000"
> Setting Configuration 1...
> Claiming USB DFU Interface...
> Setting Alternate Setting ...
> Determining device status: state = dfuIDLE, status = 0
> dfuIDLE, continuing
> Transfer Size = 0x0400
> bytes_per_hash=280
> Starting download: [##################################################] finished!
> state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
> Done!
> Resetting USB to switch back to runtime mode

**The content of `platformio.ini`:**
```
> [env:maple_mini_b20]
platform = ststm32
framework = arduino
board = maple_mini_b20
```

**Source file to reproduce issue:**

* Arduino Framework example _arduino-blink_ (`Blink.cpp`)

### Additional info

* problem was observed also on a Linux operating system (Debian Stretch)
* an other ST STM32F1 board (_blue pill_) is also affected
* problem doesn't occur when using the [Arduino-IDE](https://github.com/rogerclarkmelbourne/Arduino_STM32) in otherwise identical conditions
* problem seems to be related to producing the ELF module or upstream steps (I replaced the maple_upload script and dfu-util)

ubis, 2018-07-29:
This could happen due wrong pull-up resistor on usb D+. More info [here](http://wiki.stm32duino.com/index.php?title=Blue_Pill). Once I've changed it, I didn't got any upload problems. Can't reproduce this issue with maple mini 2.0 either.

mr-freeze, 2018-07-29:
Thank you very much for your help. Yes, the blue pill board has the problematic resistor value (concerning the maple mini clone I'm not sure). But I do not think that would explain the behavior (Arduino IDE flash process being rock-solid). If I find out something new, I will keep you informed.

ivankravets, 2018-08-01:
@ubis could we update Arduino STM32 to the latest version? Do we need some changes to build script?

ubis, 2018-08-01:
@mr-freeze That's great, seems like it is indeed software issue.
@ivankravets Yeah it seems. Board definitions weren't changed, so we're should be good.

mr-freeze, 2018-08-09:
@ivankravets, PFA my [test-log.txt](https://github.com/platformio/platform-ststm32/files/2274533/test-log.txt).

Issue has been fixed in the the upstream ST STM32 platform version. Thank you very much for your support.

I did the following to prove the effectiveness of the fix:
1. create configuration in [platformio.ini](https://github.com/platformio/platform-ststm32/files/2274526/platformio.ini.txt) with two environments for the two platform versions (the upstream one is named 
`issue_126`)
2. uninstall ststm32 (Arduino framework) package
3. install the STM32duino-bootloader in the maple mini clone board
4. clean the environment
5. build and upload firmware twice to reproduce the defect
6. install the STM32duino-bootloader again
7. build and upload firmware twice in the new environment successfully!

hint:
* PlatformIO installs the required packages automatically (but you know that better than me ;-)
* the second try uses the verbose option of pio run

Version can be released.

ivankravets, 2018-08-09:
Thanks! New release is out. Please update.

commit 3e3e76cf (2018-08-09): Update STM32Duino Core to #d4b3cd114ba567dbd917b7373f2160d14dd29fd4 // Resolve #126