platformio/platform-ststm32 #140: OpenCR Board Support (STM32F746ZGT6)

airhorns, 2018-07-30:
It'd be awesome to be able to work on OpenCR robotics projects with PlatformIO instead of the Arduino IDE. OpenCR is an awesome project: it has a wide variety of inputs and outputs, Arduino UNO compatible pinouts for shields, integrated power management for input of 5-24V with shore power / battery uninterrupted support, and really isn't too much money. It's the board that does the real time stuff for Turtlebot3 if you are familiar with it.

ROBOTIS has already done a bunch of work to get a toolchain in place for loading code onto the thing, so I hope that it's just a case of configuring PlatformIO to know what to put where! 

The documentation for the board can be found here: http://emanual.robotis.com/docs/en/parts/controller/opencr10/

And the manifest for the Arduino IDE Board Manager can be found here: https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCR/master/arduino/opencr_release/package_opencr_index.json

and the code for the existing OpenCR TurtleBot3 firmware can be found here: https://github.com/ROBOTIS-GIT/opencr

ivankravets, 2019-02-21:
Could you try this board manifest https://github.com/platformio/platform-ststm32/blob/develop/boards/nucleo_f746zg.json
Does it work?

marcosjl, 2019-10-15:
Hi !
I jump in as I have tried to use the nucleo_f746zg manifest... Did not succeeded : cross compilation works fine, but I'm just enable to upload the code to the openCR board...
Jose

marcosjl, 2019-10-23:
Arduino framework provided by Robotis uses a specific bootloader as in platformio, boot loader defined for STM32 boards seems to be openCD.
Is there a description of what needs to be done to define a specific board / bootloader in the doc ? I've found this : https://docs.platformio.org/en/latest/platforms/custom_platform_and_board.html, but I did not manage to have something up & running...
Thanks in advance

valeros, 2019-10-23:
Hi @marcosjl ! What upload method are you using? Looks like the first 0x40000 bytes are reserved for bootloader. You could try to specify this offset in `platformio.ini`:
```
[env:nucleo_f746zg]
platform = ststm32
board = nucleo_f746zg
framework = arduino
board_upload.offset_address=0x08040000
```

marcosjl, 2019-10-30:
Hi @valeros !
Nope it does not work either. 

Trying to compile a basic empty program (no code lines in the mbed template for main.cpp).

I get this error messages on the upload stage : 

`Configuring upload protocol...
AVAILABLE: blackmagic, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
Uploading .pio\build\nucleo_f756zg\firmware.bin
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

srst_only separate srst_nogate srst_open_drain connect_deassert_srst

Error: open failed
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

*** [upload] Error 1
`

cguimaraes, 2022-02-09:
The previous PR (#597) provides the required support for OpenCR in PlatformIO.
However, it is also required to include the following core files as a new package:
  https://github.com/cguimaraes/framework-arduinoststm32-opencr

@valeros: any advice on how to make piopm aware of it and automatically fetch it?
So far I have to install it manually, as we described in the bottom of this blog post: https://zenoh.io/blog/2022-02-08-dragonbot/

concavegit, 2022-03-05:
Thanks for sharing the package, it works for me and I am glad I can use IDEs suited for larger projects now. For anyone else following the blog post, I also needed to add

```
upload_command = ${platformio.packages_dir}/framework-arduinoststm32-opencr/tools/linux/opencr_ld /dev/ttyACM* 115200 $SOURCE 1
```

to my platform.ini to upload to my OpenCR board.

cguimaraes, 2022-03-06:
Great that it worked without issues and that it is being useful @concavegit .
And thank you for providing the upload command, as I forgot to indicate it in the steps above.