platformio/platform-ststm32 #141: mbed: CAN peripheral not enabled on bluepill_f103c8 / genericSTM32F103C8:

cesarvandevelde, 2018-08-02:
The following code example does not compile on `bluepill_f103c8` / `genericSTM32F103C8`, but does compile on `nucleo_f103rb`:

```C++
#include <mbed.h>
#include <CAN.h>

CAN can(PB_8, PB_9);

int main() {
  while (true) {

  }

  return 0;
}
```

```ini
[env:stm32]
platform = ststm32
framework = mbed
board = bluepill_f103c8
; board = genericSTM32F103C8
; board = nucleo_f103rb
```

There should not be a difference, as the only difference between the F103C8 and the F103RB is that the latter has more flash memory. The root cause is that the `bluepill_f103c8` target is missing a couple of config flags in `targets.json`. I've opened an issue describing this behavior here: ARMmbed/mbed-os#7654 . 

However, simply adding `"device_has_add": ["CAN", "SERIAL_FC", "SERIAL_ASYNCH", "FLASH"],` to `targets.json` does not solve the issue on PlatformIO. It seems that PlatformIO uses a more complex build process (with variants remapping etc), and I can't seem to find the source repository for the `framework-mbed` package on GitHub, hence the second issue here. 

How can I fix the PlatformIO side of this issue?

ivankravets, 2018-08-08:
I think we can close this issee? When mbed merges your PR, the changes will be reflected in the next their update and in PlatformIO too.

cesarvandevelde, 2018-08-09:
Hi @ivankravets, you're right about the changes to `targets.json`. This fix will propagate through PlatformIO once PR is merged. However, while investigating, I found an additional problem related to these features in the PlatformIO mbed toolchain. That's why I'd like to keep this issue open for now...

The `piombedgenerator` script is missing a bunch of mbed libraries. For instance, the `nvstore` library is missing from the list: 
```Python
MBED_LIBS = [
    join(ROOT, "rtos"),
    join(ROOT, "events"),
    join(ROOT, "features", "mbedtls"),
    join(ROOT, "features", "netsocket"),
    join(ROOT, "features", "filesystem"),
    join(ROOT, "features", "unsupported", "dsp"),
    join(ROOT, "features", "unsupported", "rpc"),
    join(ROOT, "features", "unsupported", "USBDevice"),
    join(ROOT, "features", "unsupported", "USBHost")
]
```

I'd like to fix this, but I can't find any source repositories for `framework-*` and `tool-*` packages. Any pointers?

ivankravets, 2018-12-11:
Could you give a link to `nvstore` library?

cesarvandevelde, 2018-12-11:
NVStore is one of the storage APIs included in mbed. This one in particular uses flash sectors to store non-volatile data.
https://os.mbed.com/docs/v5.8/reference/nvstore.html