platformio/platform-ststm32 #143: framework-cmsis depends on framework-mbed

tve, 2018-09-03:
The framework-cmsis comes with only a very small number of linker scripts. But it does try to match the board's `variant` with the linker scripts included in framework-mbed, see https://github.com/platformio/platform-ststm32/blob/develop/builder/frameworks/cmsis.py#L64
It took me a while to figure this out and I would have appreciated if either: framework-cmsis depends on framework-mbed and thus pio automatically installs the latter, or if the documentation of framework-cmsis prominently described this dependency.

ivankravets, 2018-10-13:
@valeros could we make CMSIS independent?

ivankravets, 2018-12-11:
Resolved in https://github.com/platformio/platform-ststm32/commit/5532cfe44c6aecf29cf2db0efc922ac56f857e27

closed 2018-12-11 by ivankravets