platformio/platform-ststm32 #148: mbed-rtos example gives warnings

Navis-Raven, 2018-09-11:
Configuration:

```ini
[env:disco_l475vg_iot01a]
platform = ststm32
board = disco_l475vg_iot01a
framework = mbed
debug_tool = stlink
build_flags = -D PIO_FRAMEWORK_MBED_RTOS_PRESENT
```

[image]

ivankravets, 2019-02-21:
Could you reproduce with the latest release?

Navis-Raven, 2019-02-23:
@ivankravets 
Hi I can't make any test, I have many serious bugs since the last updates 

Look at here: https://github.com/platformio/platformio-core/issues/2159

ivankravets, 2019-02-23:
Could you try to remove `C\.platformio` folder and restart VSCode? It seems that you have non-ASCII chars in user name.

Navis-Raven, 2019-03-06:
I ll make some tests now

Navis-Raven, 2019-03-06:
@ivankravets 
that's it the warnings has little evolved but are still here

[image]

```
src\main.cpp: In function 'void queue_thread(const void*)':
src\main.cpp:16:17: warning: 'static osStatus rtos::Thread::wait(uint32_t)' is deprecated: Static methods only affecting current thread cause confusion. Replaced by ThisThread::sleep_for. [since mbed-os-5.10] [-Wdeprecated-declarations]
Thread::wait(1000);
^~~~
In file included from C:/.platformio/packages/framework-mbed/rtos/rtos.h:30,
from C:/.platformio/packages/framework-mbed/mbed.h:22,
from src\main.cpp:1:
C:/.platformio/packages/framework-mbed/rtos/Thread.h:442:21: note: declared here
static osStatus wait(uint32_t millisec);
^~~~
src\main.cpp:16:26: warning: 'static osStatus rtos::Thread::wait(uint32_t)' is deprecated: Static methods only affecting current thread cause confusion. Replaced by ThisThread::sleep_for. [since mbed-os-5.10] [-Wdeprecated-declarations]
Thread::wait(1000);
^
In file included from C:/.platformio/packages/framework-mbed/rtos/rtos.h:30,
from C:/.platformio/packages/framework-mbed/mbed.h:22,
from src\main.cpp:1:
C:/.platformio/packages/framework-mbed/rtos/Thread.h:442:21: note: declared here
static osStatus wait(uint32_t millisec);
^~~~
src\main.cpp:16:26: warning: 'static osStatus rtos::Thread::wait(uint32_t)' is deprecated: Static methods only affecting current thread cause confusion. Replaced by ThisThread::sleep_for. [since mbed-os-5.10] [-Wdeprecated-declarations]
Thread::wait(1000);
^
In file included from C:/.platformio/packages/framework-mbed/rtos/rtos.h:30,
from C:/.platformio/packages/framework-mbed/mbed.h:22,
from src\main.cpp:1:
C:/.platformio/packages/framework-mbed/rtos/Thread.h:442:21: note: declared here
static osStatus wait(uint32_t millisec);
^~~~
src\main.cpp: In function 'int main()':
src\main.cpp:21:31: warning: 'rtos::Thread::Thread(void (*)(const void*), void*, osPriority_t, uint32_t, unsigned char*)' is deprecated: Thread-spawning constructors hide errors. Replaced by thread.start(callback(task, argument)). [since mbed-os-5.1] [-Wdeprecated-declarations]
Thread thread(queue_thread);
^
In file included from C:/.platformio/packages/framework-mbed/rtos/rtos.h:30,
from C:/.platformio/packages/framework-mbed/mbed.h:22,
from src\main.cpp:1:
C:/.platformio/packages/framework-mbed/rtos/Thread.h:245:5: note: declared here
Thread(void (*task)(void const *argument), void *argument = NULL,
^~~~~~
src\main.cpp:29:20: warning: format '%x' expects argument of type 'unsigned int', but argument 2 has type 'osStatus' {aka 'long int'} [-Wformat=]
printf("queue->get() returned %02x status\n\r", evt.status);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~~
src\main.cpp:31:20: warning: format '%d' expects argument of type 'int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Wformat=]
printf("queue->get() returned %d\n\r", evt.value.v);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  ~~~~~~~~~~~
```

Navis-Raven, 2019-03-06:
+ these ones

[image]

[image]

Navis-Raven, 2019-03-06:
@ivankravets  in facts there's more warnings since last time, and is due to the example mbed-rtos code which is not up to date with the new arm-mbed-rtos standards. Please consider these warning to update the example code(s)

[image]

Navis-Raven, 2019-03-06:
Copy of the warnings:

{
	"resource": "/c:/Users/Solid Snake/Documents/PlatformIO/Projects/190306-071507-mbed-rtos/src/main.cpp",
	"owner": "cpp",
	"severity": 4,
	"message": "'static osStatus rtos::Thread::wait(uint32_t)' is deprecated: Static methods only affecting current thread cause confusion. Replaced by ThisThread::sleep_for. [since mbed-os-5.10] [-Wdeprecated-declarations]",
	"startLineNumber": 16,
	"startColumn": 17,
	"endLineNumber": 16,
	"endColumn": 17
}