platformio/platform-ststm32 #156: mbedos 5.10.x with ESP8266 fails to build

mburumaxwell, 2018-11-04:
I am facing an error using ESP8266 in mbed on platformio.
Steps to reproduce:

1. create project based on any example (best case is _mbed-rtos-ethernet_)
2. build project (it succeeds)
3. Add `#include "ESP8266Interface.h"`
4. build project (it fails)

Errors are too many but they all point towards missing header files like WiFiAccessPoint.h, WiFiInterface .h etc.
I beleive this is probably an issue with mbedos itself but on the mbed compiler it works. Is there a way to resolve this issue?

ivankravets, 2019-02-21:
Does it work with the latest release?

mburumaxwell, 2019-03-04:
with support for `mbed_app.json` this should definitely work. I moved to a pure `mbed-cli` experience and custom debugging tools so I have not been able to test. However, I can comfortably close the issue.

closed 2019-03-04 by mburumaxwell