platformio/platform-ststm32 #160: How to add support to STM32F105?

sinosoidal, 2018-11-10:
Hi,

How can I add support to the STM32F105? What are the necessary steps. Am I able to do it by myself?

I was tinkering with board and env bluepill_f103c8 and basic GPIO was working but as soon as I tried to work with UART I have stumbled in problems.

Thanks in advance!

valeros, 2018-12-11:
As I can see the only framework you can use with `F105` series is `stm32cube`, but you'll need to manually add your [custom board](http://docs.platformio.org/en/latest/platforms/creating_board.html) to the ststm32 platform.

brightproject, 2024-12-20:
Good day @sinosoidal and @valeros 🙂
I have a board with a `STM32F105RCT6` microcontroller

[image]

Making some attempts to add it to `.pio`
For this, I created a `genericSTM32F105RC.json` file with the following contents

```
{
  "build": {
    "core": "stm32",
    "cpu": "cortex-m3",
    "extra_flags": "-DSTM32F105xC -DSTM32F1",
    "f_cpu": "72000000L",
    "hwids": [
      [
        "0x1EAF",
        "0x0003"
      ],
      [
        "0x1EAF",
        "0x0004"
      ]
    ],
    "mcu": "stm32f105rct6",
    "product_line": "STM32F105xC",
    "variant": "STM32F1xx/F105R(8-B-C)T"
  },
  "debug": {
    "jlink_device": "STM32F105RC",
    "openocd_target": "stm32f1x",
    "svd_path": "STM32F107.svd"
  },
  "frameworks": [
    "arduino",
    "cmsis",
    "libopencm3",
    "stm32cube"
  ],
  "name": "STM32F105RC (64k RAM. 256k Flash)",
  "upload": {
    "disable_flushing": false,
    "maximum_ram_size": 65536,
    "maximum_size": 262144,
    "protocol": "stlink",
    "protocols": [
      "jlink",
      "cmsis-dap",
      "stlink",
      "blackmagic",
      "serial",
      "dfu"
    ],
    "require_upload_port": true,
    "use_1200bps_touch": false,
    "wait_for_upload_port": false
  },
  "url": "https://www.st.com/en/microcontrollers-microprocessors/stm32f105rc.html",
  "vendor": "Generic"
}

```

Data was partially taken from the `boards_entry.txt` file, which is located in the archive below.

The `JSON` file was placed in the folder

> .platformio\platforms\ststm32\boards

Further in the folder

> .platformio\packages\framework-arduinoststm32\variants\STM32F1xx

I found a variant for `F105R(8-B-C)T` and added there the linker `ldscript.ld` from the board `stm32f103`
Archive attached:

[F105R(8-B-C)T.zip](https://github.com/user-attachments/files/18209292/F105R.8-B-C.T.zip)

`platformio.ini` is like this:

> [platformio]
> description = Testing board on STM32F105RCT6
> default_envs = STM32F105RCT6
> 
> [env:STM32F105RCT6]
> platform = ststm32
> board = genericSTM32F105RC
> framework = arduino

```
#include <Arduino.h>

#define LED_UART1 (PA8)

#define BAUD_RATE 115200 // bitrate or baudrate ~ 100 kHz

#define UART1_SERIAL_TX (PA9)
#define UART1_SERIAL_RX (PA10)

#define UART2_SERIAL_TX (PC10)
#define UART2_SERIAL_RX (PC11)

// Initialize serial port
HardwareSerial PORT1(USART1);
HardwareSerial PORT2(UART4);

void setup() {
 
    // initialize the pin to output
    pinMode (LED_UART1, OUTPUT);
    
    // set level on pin to 0
    digitalWrite (LED_UART1, LOW);
    
    // Start the serial port for monitoring and data transfer
    PORT1.setTx(UART1_SERIAL_TX);
    PORT1.setRx(UART1_SERIAL_RX);
    PORT1.begin(BAUD_RATE); // PA9, PA10
    
    PORT2.setTx(UART2_SERIAL_TX);
    PORT2.setRx(UART2_SERIAL_RX);
    PORT2.begin(BAUD_RATE); // PC10, PC11
    
    PORT1.println("UART1");
    PORT2.println("UART4");

} // SETUP

void loop() {

    digitalWrite(LED_UART1, HIGH);
    delay(200); 
    digitalWrite(LED_UART1, LOW);
    delay(200);  

} // LOOP
```

The code compiles without errors, there is a warning about `SystemClock_Config`

> Compiling .pio\build\STM32F105RCT6\FrameworkArduinoVariant\generic_clock.c.o
> .platformio\packages\framework-arduinoststm32\variants\STM32F1xx\F105R(8-B-C)T\generic_clock.c: In function 'SystemClock_Config':
> .platformio\packages\framework-arduinoststm32\variants\STM32F1xx\F105R(8-B-C)T\generic_clock.c:24:2: warning: #warning "SystemClock_Config() is empty. Default clock at reset is used." [-Wcpp]
>  24 | #warning "SystemClock_Config() is empty. Default clock at reset is used."

At first I didn't give much attention to the warning, because the firmware controlled the `LED` and `Serial` gave out data, but when I needed to output data to `UART2` or `UART4` and use the `CAN` or `MODBUS` I realized that they were not working.
Then I looked at the `generic_clock.c` file and saw, that it is empty.

```
WEAK void SystemClock_Config(void)
{
  /* SystemClock_Config can be generated by STM32CubeMX */
#warning "SystemClock_Config() is empty. Default clock at reset is used."
}
```

I also understood from the datasheet that an internal generator is needed for the pins and `UART1`, in order for other peripherals to work, an external quartz generator must be configured.

[image]

Tried to copy the contents of the `generic_clock.c` file from other boards, but it didn't help.
Also studied the material from another framework

> https://github.com/stm32duino/Arduino_Core_STM32/wiki/Add-a-new-variant-(board)#4---system-clock-configuration

But it didn't help either and would appreciate any help.

brightproject, 2024-12-24:
Put this code in here

> .platformio\packages\framework-arduinoststm32\variants\STM32F1xx\F105R(8-B-C)T\generic_clock.c

```
/*
 *******************************************************************************
 * Copyright (c) 2020-2021, STMicroelectronics
 * All rights reserved.
 *
 * This software component is licensed by ST under BSD 3-Clause license,
 * the "License"; You may not use this file except in compliance with the
 * License. You may obtain a copy of the License at:
 *                        opensource.org/licenses/BSD-3-Clause
 *
 *******************************************************************************
 */
#if defined(ARDUINO_GENERIC_F105R8TX) || defined(ARDUINO_GENERIC_F105RBTX) || defined(ARDUINO_GENERIC_F105RCTX)
#include "pins_arduino.h"

/**
  * @brief  System Clock Configuration
  * @param  None
  * @retval None
  */
// WEAK void SystemClock_Config(void)
// {
//   /* SystemClock_Config can be generated by STM32CubeMX */
// #warning "SystemClock_Config() is empty. Default clock at reset is used."
// }

WEAK void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

  /** Initializes the RCC Oscillators according to the specified parameters
  * in the RCC_OscInitTypeDef structure.
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_NONE;
  RCC_OscInitStruct.PLL2.PLL2State = RCC_PLL_NONE;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }

  /** Initializes the CPU, AHB and APB buses clocks
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_HSI;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_0) != HAL_OK)
  {
    Error_Handler();
  }

  /** Configure the Systick interrupt time
  */
  __HAL_RCC_PLLI2S_ENABLE();
}

#endif /* ARDUINO_GENERIC_* */

```