platformio/platform-ststm32 #164: Can not use dfu for upload on Windows 10, VS Code and PlatformIO

dankar, 2018-12-07:
I've tried almost all combinations of target boards and platforms (but my main interest is in generic STM32F407VGT6 with STM32Cube). I am using Windows 10 with VS Code and Platform IO.

If I don't choose an upload protocol the console window says: 
"Configuring upload protocol...
AVAILABLE: dfu, jlink, stlink"

Then it fails since stlink is default and I only have a dfu device available.

After I set "upload_protocol = dfu" in platformio.ini this is the output:

```
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F407VGT6.html
PLATFORM: ST STM32 > STM32F407VG (192k RAM. 1024k Flash)
HARDWARE: STM32F407VGT6 168MHz 192KB RAM (1MB Flash)
DEBUG: CURRENT(jlink) EXTERNAL(jlink, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
No dependencies
Warning! Unknown upload protocol dfu
Checking size .pioenvs\genericSTM32F407VGT6\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   0.0% (used 44 bytes from 196608 bytes)
PROGRAM: [          ]   0.1% (used 1264 bytes from 1048576 bytes)
============================================================================================== [SUCCESS] Took 3.05 seconds ==============================================================================================
```

The relevant line is "Warning! Unknown upload protocol dfu" which I can't find any information about.

Regards,
Daniel

coyt, 2018-12-10:
I'm adding to your post first because I'm fighting the same problem - i'll open a new issue if no one responds here. I think I'm a bit further down this problem than you - and I did get some stuff working - so here it goes:

1. I'm trying to write code, as you are, for generic cheap STM32F407VGT6 dev board. I also want to use the stm32cube framework, and I want to upload code directly from platformio with the USB DFU bootloader pre-burned on the F407 by STM. 

2. PlatformIO website claims that for  "genericSTM32F407VET6" you can use stlink OR dfu. THIS IS TOTALLY WRONG currently. 

[image]

3. As far as I can tell, platformio tries to use the tool "DFU-UTIL" to upload code to MCU's such as the STM32F103 blue pill boards. I could not get this program to work with STM's pre-burned USB DFU bootloader which came with my MCU. I got the same error as you. 

[image]

[image]

4. When my dev board is connected to pc (Win 10) it does show up as a STM DFU USB device - I tried many things - like changing the drivers using Zadig to try and make this STM DFU bootloader work with "DFU-UTIL" but i failed. 

5. Eventually I downloaded STM's "DfuSe" program suite from the [main STM32F407VE webpage.](https://www.st.com/en/microcontrollers/stm32f407ve.html). Install the program. 

6. While platformIO didn't upload, it should have compiled your program and made a firmware.bin for your STM32 somewhere in your project folder previously - find this .bin file and convert it to a .dfu file using the "DfuFileMgr" program included in the "DfuSe" suite you just installed. There are a few numbers you need to get right. 

6a: Set Vendor Id to: "0483"
6b: Set Product Id to: "DF11"
6c: Set Version to: "2200"
6d: During multi-bin injection: set the address field to: 08000000 so your .bin starts at the right place. 

7. Finally, connect your board in boot loader mode to pc. I had to set boot 0 jumper to 3V3 and hold reset as I connected usb to the pc. Now you can use "DfuSeDemo" program to flash the .dfu file you made onto your board. Should work!

PlatformIO devs - any way to get this process fixed or integrated into the VScode plugin? It's kinda a PITA to do manually. Maybe at least fix the documentation so people don't expect DFU to work out of box? Thanks!

And I'm gonna keep looking into this... it looks like there's a python file "main.py" under ./platformio/platforms/ststm32xxxxxxxx/builder/ containing a big elif regarding upload protocols. Seems "serial" and "dfu" are defined here? Can calls to use STM's "DfuSe" programming suite be added here as an option for this board in order to automate the process I describe above? Alternatively, can DFU-UTIL be setup to work with the boot-loader inherent to the F407? Or can a new boot-loader be made for this chip?

coyt, 2018-12-11:
No it does not appear to work. I just tried quickly but I'll try again later when I can take a closer look.

Again, I see you referenced an issue concerning the STM32F103 chip. I'm not familiar with the issue referenced in that issue #155, but I've personally had success with the 103 chip before. 

Not the case with the 407:

The DFU bootloader included in the STM32F407VET6 by STM I believe is fundamentally different and requires a fundamentally different DFU upload tool to work. Can you confirm if this is correct?

ivankravets, 2018-12-11:
>  Can you confirm if this is correct?

@ubis could you comment here?

noisymime, 2019-03-07:
The step that's missing currently is the conversion of the .hex/.bin file into a .dfu. The dfu-util releases include a python script (https://sourceforge.net/p/dfu-util/dfu-util/ci/master/tree/dfuse-pack.py) that can take care of this conversion, including adding all the required vendor/product IDs etc. This method is probably better than attempting to use the ST DfuSe utility, which is both non-free and for Windows only. 

Once that conversion is performed, then the resulting .dfu file can be uploaded using dfu-util as per normal to any f407 device.

Would it be possible to add the `dfuse-pack.py` script and call it as part of the upload process for f407 boards?

rijesha, 2019-04-30:
The dfu upload tool included with platformIO is the one that comes with stm32duino. This upload tool is expecting the Maple bootloader.(the stm32f1 chips didn't have a built in usb bootloader) To use the tool included with platformIO you will first need to flash Maple bootloader onto your stm32f4.

Alternatively you can use the built in usb bootloader and dfu_util. You do not need to convert the bin to a .dfu if you are using the stm32f4*. Check out this link for more info on the topic. http://dfu-util.sourceforge.net/dfuse.html

I have an example project that sets the UPLOADCMD var and uploads using dfu_util.
https://github.com/rijesha/stm32f4_dfu_and_virtual_com_port_platform_io

Checkout README.md, platformio.ini and dfu_upload.py for more details on how it is implemented.

valeros, 2019-07-17:
Should be fixed in the development branch. Now `dfu-util` is used with `dfu` upload protocol (except for stm32f1x based boards)

commit 304c95dd (2018-12-11): Allow DFU protocol for other frameworks // Resolve #164
closed 2019-07-18 by valeros