platformio/platform-ststm32 #166: Port board configuration

masoudr, 2018-12-14:
Hi,
I'm trying to port my project in PIO to original arm's mbed framework with `mbed-cli`. The code compiles fine but it runs into a problem on executing because using large arrays which worked fine on PIO but have some problem with mbed framework. I search and found that PIO uses some unique configurations on each target and on the other side mbed uses `mbed_app.json` and `mbed_config.h` for the configuration. I was wondering if there are any public way that I can port the PIO configuration for my target?
I'm using DISCO_F407VG target.
Thanks.

ivankravets, 2018-12-14:
Could you share your `mbed_app.json` and mbed_config.h?

ivankravets, 2018-12-14:
Thanks! Could you try the next?  See https://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-flags

```
[env:myenv]
platform = ststm32
framework = mbed
board = ...
build_flags =
  -D MBED_CONF_PLATFORM_STDIO_BAUD_RATE=115200
  -D MBED_CONF_TARGET_STDIO_UART_RX=PA_10
  -D MBED_CONF_TARGET_STDIO_UART_TX=PA_9
```

Please clean and rebuild the project. Does it work? If yes, we will implement automatic handling of `mbed_app.json`

masoudr, 2018-12-14:
Thanks, it worked for me but I set `MBED_CONF_PLATFORM_DEFAULT_SERIAL_BAUD_RATE` for the default baud rate. I found the problem, it seems that mbed uses low `MBED_CONF_APP_MAIN_STACK_SIZE` (4096) which made problems for me, but I was wondering what value set in PIO, I found the default `mbed_config.h` and realize that it is not set in there. So I resized it in `mbed_app.json`:
```
{
    "config": {
        "main-stack-size": {
            "value": 10240
        }
    },
    "target_overrides": {
        "*": {
            "platform.stdio-baud-rate": 115200,
            "target.stdio_uart_tx": "PA_9",
            "target.stdio_uart_rx": "PA_10"
        }
    }
}
```

ivankravets, 2019-02-21:
We added official support for `mbed_app.json` in the latest release. See docs https://docs.platformio.org/en/latest/frameworks/mbed.html#the-configuration-system

closed 2019-02-21 by ivankravets