platformio/platform-ststm32 #167: Upload problem with ST-Link

masoudr, 2018-12-14:
Hi,
I updated to the latest version of PIO. Now when I want to upload to the board I get the following error. I haven't faced this problem on older releases. I'm using ST-Link V2.0 with SWD connection.
config file:
```
[env:disco_f407vg]
platform = ststm32
board = disco_f407vg
framework = mbed
upload_protocol = stlink
```

> GNU MCU Eclipse 64-bits Open On-Chip Debugger 0.10.0+dev-00392-gbe9ef0b0 (2018-01-12-15:03)
> Licensed under GNU GPL v2
> For bug reports, read
> http://openocd.org/doc/doxygen/bugs.html
> Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
> adapter speed: 2000 kHz
> adapter_nsrst_delay: 100
> none separate
> srst_only separate srst_nogate srst_open_drain connect_deassert_srst
> Info : Unable to match requested speed 2000 kHz, using 1800 kHz
> Info : Unable to match requested speed 2000 kHz, using 1800 kHz
> Info : clock speed 1800 kHz
> Info : STLINK v2 JTAG v29 API v2 SWIM v7 VID 0x0483 PID 0x3748
> Info : using stlink api v2
> Info : Target voltage: 3.247525
> Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
> Info : Listening on port 3333 for gdb connections
> Info : Unable to match requested speed 2000 kHz, using 1800 kHz
> Info : Unable to match requested speed 2000 kHz, using 1800 kHz
> adapter speed: 1800 kHz
> Error: timed out while waiting for target halted
> *** [upload] Error 884

ivankravets, 2018-12-14:
Could you try to reconnect board? Change USB?

ivankravets, 2018-12-14:
Do you use external ST-Link?

masoudr, 2018-12-14:
@ivankravets Yes, I am.

ivankravets, 2018-12-14:
Did you wire RST/nRST?

ivankravets, 2018-12-14:
Thanks! Now I understand. You need to declare a custom upload command. See https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#custom-upload-tool

What do I recommend?

1. Try to do uploading in verbose mode via `pio run -t upload -v`
2. Copy full command where openOCD is used
3. Use this command for 
```python
env.Replace(
    UPLOADCMD="executable -arg1 -arg2 $SOURCE"
)
```

See also https://github.com/platformio/platform-ststm32/blob/develop/builder/main.py#L192:L204

masoudr, 2018-12-15:
Thanks, I tried that but I get `Warning! Unknown upload protocol custom` in pio output here is my configs:
`platformio.ini`:
```
[env:disco_f407vg]
platform = ststm32
board = disco_f407vg
framework = mbed

extra_scripts = extra_script.py
upload_protocol = custom
; each flag in a new line
upload_flags =
    -f \"scripts/interface/stlink.cfg\"
    -f \"scripts/target/stm32f4x.cfg\"

```
`extra_script.py`:
```
Import("env")

# please keep $SOURCE variable, it will be replaced with a path to firmware

# In-line command with arguments
env.Replace(
    UPLOADCMD="openocd -arg1 -arg2 -c \"program {{$SOURCE}} verify reset exit\""
)

```

closed 2018-12-21 by ivankravets

ivankravets, 2018-12-21:
See example with a custom command for openOCD in the docs https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#custom-upload-tool

masoudr, 2018-12-21:
@ivankravets Thanks using extra scripts didn't work for me and I got some mismatch on verifying, but using default upload_protocol works.

masoudr, 2019-01-29:
Recently I tried to upload but I get this error with `custom upload` like in the example.
> Open On-Chip Debugger 0.10.0 (2018-07-28) [https://github.com/sysprogs/openocd]
> Licensed under GNU GPL v2
> For bug reports, read
> http://openocd.org/doc/doxygen/bugs.html
> Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
> Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
> adapter speed: 2000 kHz
> adapter_nsrst_delay: 100
> none separate
> Info : Unable to match requested speed 2000 kHz, using 1800 kHz
> Info : Unable to match requested speed 2000 kHz, using 1800 kHz
> Info : clock speed 1800 kHz
> Info : STLINK v2 JTAG v29 API v2 SWIM v7 VID 0x0483 PID 0x3748
> Info : using stlink api v2
> Info : Target voltage: 3.203137
> Info : stm32f4x.cpu: hardware has 6 breakpoints, 4 watchpoints
> Info : Listening on port 3333 for gdb connections
> Info : Unable to match requested speed 2000 kHz, using 1800 kHz
> Info : Unable to match requested speed 2000 kHz, using 1800 kHz
> adapter speed: 1800 kHz
> target halted due to debug-request, current mode: Thread
> xPSR: 0x01000000 pc: 0x08002db8 msp: 0x20020000
> Info : Unable to match requested speed 8000 kHz, using 4000 kHz
> Info : Unable to match requested speed 8000 kHz, using 4000 kHz
> adapter speed: 4000 kHz
> ** Programming Started **
> auto erase enabled
> Info : device id = 0x100f6413
> Info : flash size = 1024kbytes
> Warn : no flash bank found for address 0x00000000
> wrote 0 bytes from file .pioenvs\disco_f407vg\firmware.bin in 0.001998s (0.000 KiB/s)
> ** Programming Finished **
> ** Verify Started **
> target halted due to breakpoint, current mode: Thread
> xPSR: 0x61000000 pc: 0x2000002e msp: 0x20020000
> Error: checksum mismatch - attempting binary compare
> diff 0 address 0x00000004. Was 0xb9 instead of 0x69
> diff 1 address 0x00000005. Was 0x2d instead of 0x1c
> diff 2 address 0x00000008. Was 0x05 instead of 0xb5
> diff 3 address 0x00000009. Was 0x2e instead of 0x1c
> diff 4 address 0x0000000c. Was 0x15 instead of 0x45
> diff 5 address 0x0000000d. Was 0x70 instead of 0x46
> ...
> diff 126 address 0x00000114. Was 0x05 instead of 0xb5
> diff 127 address 0x00000115. Was 0x2e instead of 0x1c
> More than 128 errors, the rest are not printed.
> ** Verify Failed **
> shutdown command invoked
> 
> *** [upload] Error 1

If I use default `upload_protocol=stlink` I should edit `platform.py` file to get it working:

```
...
 server_args = [
                        "-f",
                        "scripts/interface/%s.cfg" % link, "-c",
                        "transport select %s" % (
                            "hla_swd"if link == "stlink" else "swd"),
                        "-f",
                        "scripts/target/%s.cfg" % debug.get("openocd_target")
                    ]
                server_args.extend(["-c", "reset_config none"])   #added this line
                debug['tools'][link] = {
...
```
I tried to add `"-c", "reset_config none"` to `extra_script` but got the same error.

commit e743176c (2018-12-21): Fix warning about "Unknown upload protocol custom" // Resolve #167