platformio/platform-ststm32 #168: STLink upload does not work with STLink clones

kiranshila, 2018-12-17:
Alright so me and @coyt have tried this and verified the problem. We both are using ST-Link V2 clones from [Amazon](https://www.amazon.com/DAOKI-ST-Link-Programming-Emulator-Downloader/dp/B01EE4WAC8/ref=pd_lpo_vtph_147_lp_t_2?_encoding=UTF8&psc=1&refRID=7PF6NY4Y8RFB4158SR9A) and haven't had success uploading. We get errors that end in some variation of
 `Error: timed out while waiting for target halted`

We have uploaded binaries using the official STM32CubeProgrammer utility with the fake device successfully.

We have tried editing the stlink.cfg in the OpenOCD scripts folder to account for the different PID/VID pair, but we get the same error.

I just got an official STLink V2-1 in the mail today and it works fine from PIO.

ivankravets, 2019-02-21:
We've updated openOCD today. Could you try the upstream version of dev/platform? See docs https://docs.platformio.org/en/latest/platforms/espressif32.html#stable-and-upstream-versions

Does it work?

JarrettR, 2019-04-09:
What target processor, @kiranshila ?

I'm getting this same issue on an STM32F3, but not an STM32F1 Blue Pill

Blue Pill, works:

```
PS C:\Users\jrain\Documents\git\platform-ststm32\examples\libopencm3-blink> pio run --target upload -v
Processing bluepill_f103c8 (platform: ststm32; board: bluepill_f103c8; framework: libopencm3)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 > BluePill F103C8
HARDWARE: STM32F103C8T6 72MHz 20KB RAM (64KB Flash)
DEBUG: CURRENT(stlink) EXTERNAL(blackmagic, jlink, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 0 compatible libraries
FSramework incompatible library C:\Users\jrain\.platformio\lib\U8g2_ID942canning dependencies...

More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
No dependencies
MethodWrapper(["checkprogsize"], [".pioenvs\bluepill_f103c8\firmware.elf"])
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   0.0% (used 0 bytes from 20480 bytes)
PROGRAM: [          ]   1.0% (used 684 bytes from 65536 bytes)
.pioenvs\bluepill_f103c8\firmware.elf  :
section           size        addr
.text              684   134217728
.preinit_array       0   134218412
.init_array          0   134218412
.fini_array          0   134218412
.data                0   536870912
.bss                 0   536870912
.comment           126           0
.ARM.attributes     51           0
Total              861
<lambda>(["upload"], [".pioenvs\bluepill_f103c8\firmware.elf"])
AVAILABLE: blackmagic, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
openocd -s C:\Users\jrain\.platformio\packages\tool-openocd -f scripts/interface/stlink.cfg -c "transport select hla_swd" -f scripts/target/stm32f1x.cfg -c "reset_config none" -c "program {.pioenvs\bluepill_f103c8\firmware.elf} verify reset ; shutdown;"
GNU MCU Eclipse 64-bit Open On-Chip Debugger 0.10.0+dev-00462-gdd1d90111 (2019-01-18-11:42)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
none separate
Info : clock speed 1000 kHz
Info : STLINK V2J17S4 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.281374
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x0800021c msp: 0x20005000
** Programming Started **
auto erase enabled
Info : device id = 0x20036410
Info : flash size = 64kbytes
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000003a msp: 0x20005000
wrote 1024 bytes from file .pioenvs\bluepill_f103c8\firmware.elf in 0.106738s (9.369 KiB/s)
** Programming Finished **
** Verify Started **
target halted due to breakpoint, current mode: Thread
xPSR: 0x61000000 pc: 0x2000002e msp: 0x20005000
verified 684 bytes in 0.030845s (21.656 KiB/s)
** Verified OK **
[... 50 lines cut ...]
http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : clock speed 1000 kHz
Info : STLINK V2J17S4 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.221593
Info : stm32f3x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
adapter speed: 1000 kHz
Error: timed out while waiting for target halted
TARGET: stm32f3x.cpu - Not halted
in procedure 'program'

embedded:startup.tcl:480: Error: ** Unable to reset target **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 505
at file "embedded:startup.tcl", line 480
*** [upload] Error 1
================================================================================================ [ERROR] Took 3.39 seconds ================================================================================================
```

For reference, I can get the STM32F3 working with an official ST-Link with nRST connected. The clone's RST pin doesn't seem to work if used in that manner.

kiranshila, 2019-04-09:
Ours was an STM32F3, but I just trashed the clone. It's like $15 for a nucleo that has an official stlink, so to get a development platform PLUS the debugger/uploader that will just work, it was worth it.

JarrettR, 2019-04-09:
Also seems related to #167 which doesn't seem resolved, really.

I'd like to get this working :)

saierd, 2022-03-06:
I had the same problem and just created #603 for it. With that change the upload works for me when I add `upload_resetmethod = none separate` to `platformio.ini`.

The same problem also exists when debugging. I could solve it there by specifying `"openocd_extra_args": ["-c", "reset_config none separate"]` in the `debug` section of my custom board definition's json file. I am not sure how this could be specified for existing boards, though.