platformio/platform-ststm32 #169: 4.6.0 Broke code that was fine in 4.5.0

semicolonTransistor, 2018-12-23:
After updating to 4.6.0 my previously completely fine project fails to compile. I think it was caused by the switch to the "official" STM32duino from the one from rogerclarkmelbourne. There are a lot of differences between the two. It would have been nice to have made this release version 5.0.0 since compatibility was utterly broken.
Examples of incompatibility:
SPI.dmaTransfer is not implemented in the official STM32_Core but is in the rogerclark core.
WiringPinMode is not defined in the official STM32_Core, this breaks a lot of libraries.

chechu2k5, 2018-12-28:
I am facing the same problem. I worked around this (spi.dmaTransfer) by defining 
`board_build.variant = ""` in my platformio.ini. from what I can tell the commit 19c4b233094afd51ff2156d5318b52fbae2c14a6 made changes to how/where the build scripts are picked up. My work around forcibly makes the build script to use the legacy mechanism which is working me for the time being. Hopefully a better solution comes along

elmarfaber, 2019-01-16:
Other incompatibilities:
SPIClass SPI_2(2);  or SPI.setModule(2); give some error

ivankravets, 2019-02-21:
1. Please switch to an upstream version of dev/platform. See docs https://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions
2. Change Arduino core to `maple`. See https://docs.platformio.org/en/latest/platforms/ststm32.html#switching-between-arduino-cores

closed 2019-02-21 by ivankravets