platformio/platform-ststm32 #170: [FR] Add support for STM32F407ZGT boards

bipsendk, 2018-12-23:
Checking the supported boards list at https://platformio.org/boards?count=50&filter%5Bframeworks%5D=arduino&filter%5Bplatform%5D=ststm32 I figured out, that STM32F407ZGT6 boards apparently isn't supported. yet.

Multiple variants exist - but at least 2 versions would be nice to have supported:

The Acelex version on https://www.aliexpress.com/item/STM32-ARM-Cortex-M4-STM32F407ZGT6-development-board-STM32F4-core-board/32827402179.html
And the mini board: https://www.aliexpress.com/item/STM32F407-core-board-minimum-system-STM32F407ZGT6-development-board-Mini-board-Science/32892913767.html

WarlockD, 2019-02-03:
Also Olimex makes one, though its an older 2011 design.  They still sell it and have schematics available.
[https://www.olimex.com/Products/ARM/ST/STM32-P407/](url)
STM32F407VG Generic is in there, and the only real difference is extra io and  analog ports.

ivankravets, 2019-02-21:
Done in https://github.com/platformio/platform-ststm32/blob/develop/boards/black_f407zg.json

closed 2019-02-21 by ivankravets