platformio/platform-ststm32 #171: Board file for STM32L412KB

jcw, 2018-12-27:
That's the µC on the Nucleo-L412 - I copied the L432 and made a few adjustments.
Please consider adding this as `boards/nucleo_l412kb.json` to the repo:

```
{
  "build": {
    "cpu": "cortex-m4",
    "extra_flags": "-DSTM32L412xx",
    "f_cpu": "80000000L",
    "mcu": "stm32l412kbu6",
    "variant": "NUCLEO_L412KB"
  },
  "debug": {
    "default_tools": [
      "stlink"
    ],
    "jlink_device": "STM32L412KB",
    "onboard_tools": [
      "stlink"
    ],
    "openocd_target": "stm32l4x",
    "svd_path": "STM32L4x2.svd"
  },
  "frameworks": [
    "arduino",
    "mbed",
    "stm32cube"
  ],
  "name": "ST Nucleo L412KB",
  "upload": {
    "maximum_ram_size": 40960,
    "maximum_size": 131072,
    "protocol": "mbed",
    "protocols": [
      "jlink",
      "stlink",
      "blackmagic",
      "mbed"
    ]
  },
  "url": "https://developer.mbed.org/platforms/ST-Nucleo-L412KB/",
  "vendor": "ST"
}
```

I'm not yet able to upload because stlink doesn't support this new µC (yet, I hope), but the above board definition file will be needed anyway.

valeros, 2019-01-28:
Thanks for the manifest! Looks like this board doesn't support arduino and mbed frameworks at the moment, so I'll enable only stm32cube.

closed 2019-01-28 by valeros

jcw, 2020-08-18:
There's another issue with the definitions for this board. The L412 has 40 KB RAM:

```
$ st-info --probe
Found 1 stlink programmers
 serial:     303636384646353735303531373137383637303433313235
 hla-serial: "\x30\x36\x36\x38\x46\x46\x35\x37\x35\x30\x35\x31\x37\x31\x37\x38\x36\x37\x30\x34\x33\x31\x32\x35"
 flash:      131072 (pagesize: 2048)
 sram:       40960
 chipid:     0x0464
 descr:      L41x
```

But the linker script sets the RAM top to 32 KB:

```
/* Highest address of the user mode stack */ 
_estack = 0x20008000;    /* end of RAM */
/* Generate a link error if heap and stack don't fit into RAM */
_Min_Heap_Size = 0x200;      /* required amount of heap  */
_Min_Stack_Size = 0x400;     /* required amount of stack */

/* Specify the memory areas */
MEMORY
{
    RAM (xrw)      : ORIGIN = 0x20000000, LENGTH = 32K
    FLASH (rx)      : ORIGIN = 0x8000000, LENGTH = 128K
}
```

(copied from `framework-stm32cube/platformio/ldscripts/STM32L412KB_DEFAULT.ld`)

jcw, 2020-08-20:
Re that 40 vs 32 KB: looks like it has been fixed in `framework-stm32cube @ 2.0.200813`. Many thanks.

commit e782b84e (2019-01-28): Add new board nucleo_l412kb // Resolve #171