platformio/platform-ststm32 #172: STM32 for Arduino framework no longer compiles, Serial2, INPUT_ANALOG, and PWM build errors.

tailwheeldriver, 2018-12-27:
- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.
------------------------------------------------------------------

### Configuration

**Operating system**: LinuxMint 18.3

**PlatformIO Version** (`platformio --version`): version 4.0.0a1, using Atom 1.33.1

### Description of problem
I had some code for the Maple Mini (2.0 bootloader) that was compiling fine a couple weeks ago. After the latest full update (of my libraries, platforms, piocore, etc.) Nothing will compile that is unique to the leaflabs/stm32duino core library. Setting things like 
`pinMode(1, PWM);` 
throws errors of 'PWM not declared in this scope'. Serial2 and Serial3 also no longer work and build with the error:
 `.pioenvs/maple_mini_b20/src/main.cpp.o: In function 'setup':
main.cpp:(.text.setup+0x1c): undefined reference to `Serial2'
collect2: error: ld returned 1 exit status
*** [.pioenvs/maple_mini_b20/firmware.elf] Error 1`

I have written a small test just to make sure it wasn't something I broke in my own code. The issue still persists with new projects, but compiles fine with ArduinoIDE v.1.8.5

#### Steps to Reproduce

1. initialize new Maple Mini b2 project with arduino framework. 
2. code with pinMode settings for analog input or PWM output, and Serial2 or Serial3
3. build
4. pinMode errors will show with example code below, commenting those out and compiling with just `Serial2.begin(9600)` will cause a different error.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```
[env:maple_mini_b20]
platform = ststm32
board = maple_mini_b20
framework = arduino
```

**Source file to reproduce issue:**
```
#include <Arduino.h>

void setup() {
  pinMode(5, INPUT);
  pinMode(1, PWM);
  pinMode(10, INPUT_ANALOG);
  Serial2.begin(9600);

}

void loop() {
}
```
### Results of platformio run:

Processing maple_mini_b20 (platform: ststm32; board: maple_mini_b20; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via '-v, --verbose' option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/maple_mini_b20.html
PLATFORM: ST STM32 > Maple Mini Bootloader 2.0
HARDWARE: STM32F103CBT6 72MHz 20KB RAM (120KB Flash)
DEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)
￼
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 27 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs/maple_mini_b20/src/main.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:5:14: error: 'PWM' was not declared in this scope
pinMode(1, PWM);
^~~
src/main.cpp:5:14: note: suggested alternative: 'PWR'
pinMode(1, PWM);
^~~
PWR
src/main.cpp:6:15: error: 'INPUT_ANALOG' was not declared in this scope
pinMode(10, INPUT_ANALOG);
^~~~~~~~~~~~
src/main.cpp:6:15: note: suggested alternative: 'INPUT_PULLUP'
pinMode(10, INPUT_ANALOG);
^~~~~~~~~~~~
INPUT_PULLUP
*** [.pioenvs/maple_mini_b20/src/main.cpp.o] Error 1`

qu1ck, 2019-01-03:
This is because of a switch to STM's official core. See this comment for workaround:
https://github.com/platformio/platform-ststm32/issues/76#issuecomment-449624969

tailwheeldriver, 2019-01-03:
Phew, thanks for that, now I can get back to work on my project. Changing my platformio.ini to include `platform = ststm32@~4.5.0` worked great and I'm back to 4.5.0. As others have said I hope this is a temporary measure, since I now can't use any new features that may come out in future releases.