platformio/platform-ststm32 #174: Unable to use DFU mode with `bluepill_f103c8`

ashthespy, 2019-01-04:
Trying to get up and running with a bluepill and having some issues.

Info:
```
>platformio --version
PlatformIO, version 4.0.0a1

>platformio run --verbose
Processing bluepill_f103c8 (upload_protocol: dfu; platform: ststm32;  COM6; board: bluepill_f103c8; framework: arduino)
```
Have already flashed the DFU bootloader from [rogerclarkmelbourne/STM32duino-bootloader](https://github.com/rogerclarkmelbourne/STM32duino-bootloader)

However, the uploading fails with:

```
> platformio run --target upload
Processing bluepill_f103c8 (platform: ststm32; board: bluepill_f103c8; framework: arduino)
------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 > BluePill F103C8
HARDWARE: STM32F103C8T6 72MHz 20KB RAM (64KB Flash)
DEBUG: CURRENT(stlink) EXTERNAL(blackmagic, jlink, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 9 compatible libraries
Scanning dependencies...
No dependencies
Checking size .pioenvs\bluepill_f103c8\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   3.7% (used 756 bytes from 20480 bytes)
PROGRAM: [==        ]  21.0% (used 13744 bytes from 65536 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = dfu
Looking for upload port...
Use manually specified: anything
Uploading .pioenvs\bluepill_f103c8\firmware.bin
maple_loader v0.1
Resetting to bootloader via DTR pulse
Searching for DFU device [1EAF:0003]...R
eset via USB Serial Failed! Did you select the right serial port?
Assuming the board is in perpetual bootloader mode and continuing to attempt dfu programming...

Found it!

Opening USB Device 0x1eaf:0x0003...
Found Runtime: [0x1eaf:0x0003] devnum=1, cfg=0, intf=0, alt=2, name="STM32duino bootloader v1.0  Upload to Flash 0x8002000"
Setting Configuration 1...
Claiming USB DFU Interface...
Setting Alternate Setting ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
Transfer Size = 0x0400
bytes_per_hash=280
Starting download: [##################################################] finished!
error resetting after download: usb_reset: could not reset device, win error: The system cannot find the file specified.
state(8) = dfuMANIFEST-WAIT-RESET, status(0) = No error condition is present
Done!
Resetting USB to switch back to runtime mode

timeout waiting for anything serial
==================================== [SUCCESS] Took 33.35 seconds ====================================
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "C:\Users\Ash\.platformio\penv\lib\atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "C:\Users\Ash\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 375, in _finalize
    backup_reports(MPDataPusher().get_items())
  File "C:\Users\Ash\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 391, in backup_reports
    for key in params:
RuntimeError: dictionary changed size during iteration
Error in sys.exitfunc:
Traceback (most recent call last):
  File "C:\Users\Ash\.platformio\penv\lib\atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "C:\Users\Ash\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 375, in _finalize
    backup_reports(MPDataPusher().get_items())
  File "C:\Users\Ash\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 391, in backup_reports
    for key in params:
RuntimeError: dictionary changed size during iteration
```

Am I doing something wrong? Do I need a different bootloader now that `ststm32` uses a the official STM32 Arduino core?

ashthespy, 2019-01-14:
I seem to have solved my issue by switching from the official ST core to rogerclarkmelbourne's one (i.e using `platform = ststm32@~4.5.0`). Funnily however, I was subsequently able to move to the official core after the first initial flash with the old core. So maybe it was just some silly user error!

greygraphics, 2019-01-14:
Hmm, I will have to try that. Thanks for the reply

ashthespy, 2019-01-20:
Just tried with a new bluepill with `rogerclarkmelbourne`'s bootloader. 
No dice using the latest `stsm32` platform (`Version: 4.6.0`) with this dfu bootloader. It flashes once, but after that I need to re-flash the bootloader. 

Is there a recommended bootloader for the new official STArduino framework?

ivankravets, 2019-02-21:
See https://github.com/platformio/platform-ststm32/issues/169#issuecomment-466175809

closed 2019-02-21 by ivankravets

me21, 2019-02-21:
In addition, DFU support has been added you official Arduino core.