platformio/platform-ststm32 #175: Issues with used OpenOCD for STM32L0xx

martinjaeger, 2019-01-08:
The STM32L0 devices have a quite complicated dual bank flash, which creates some issues with the version of OpenOCD that comes with Platformio.

Following command generated by PlatformIO fails:
```
~/.platformio/packages/tool-openocd/bin/openocd -s /home/martin/.platformio/packages/tool-openocd -f scripts/interface/jlink.cfg -c "transport select swd" -f scripts/target/stm32l0.cfg -c "program {{.pioenvs/my_device/firmware.elf}}  verify reset; shutdown;"
GNU MCU Eclipse 64-bits Open On-Chip Debugger 0.10.0+dev-00392-gbe9ef0b0 (2018-01-12-14:56)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
swd
adapter speed: 300 kHz
adapter_nsrst_delay: 100
none separate
cortex_m reset_config sysresetreq
Info : J-Link V9 compiled Oct 25 2018 11:46:07
Info : Hardware version: 9.40
Info : VTarget = 3.298 V
Info : clock speed 300 kHz
Info : SWD DPIDR 0x0bc11477
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : Listening on port 3333 for gdb connections
adapter speed: 300 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x08001b3c msp: 0x20005000
STM32L0: Enabling HSI16
adapter speed: 2500 kHz
** Programming Started **
auto erase enabled
Info : Device: STM32L0xx (Cat.5)
Info : STM32L flash has dual banks. Bank (0) size is 96kb, base address is 0x8000000
Warn : target was in unknown state when halt was requested
Info : SWD DPIDR 0x0bc11477
Info : SWD DPIDR 0x0bc11477
Error: error writing to flash at address 0x08000000 at offset 0x00000000
embedded:startup.tcl:476: Error: ** Programming Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 532
at file "embedded:startup.tcl", line 476
```

If I use exactly the same command with the openocd provided by Arch Linux, I still get an Error, but I can successfully flash my device:
```
/usr/bin/openocd -s /home/martin/.platformio/packages/tool-openocd -f scripts/interface/jlink.cfg -c "transport select swd" -f scripts/target/stm32l0.cfg -c "program {{.pioenvs/my_device/firmware.elf}}  verify reset; shutdown;"
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
swd
adapter speed: 300 kHz
adapter_nsrst_delay: 100
none separate
cortex_m reset_config sysresetreq
Info : No device selected, using first device.
Info : J-Link V9 compiled Oct 25 2018 11:46:07
Info : Hardware version: 9.40
Info : VTarget = 3.300 V
Info : clock speed 300 kHz
Info : SWD DPIDR 0x0bc11477
Info : stm32l0.cpu: hardware has 4 breakpoints, 2 watchpoints
adapter speed: 300 kHz
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x08001b3c msp: 0x20005000
STM32L0: Enabling HSI16
adapter speed: 2500 kHz
** Programming Started **
auto erase enabled
Info : Device: STM32L0xx (Cat.5)
Info : STM32L flash has dual banks. Bank (0) size is 128kb, base address is 0x8000000
Error: stm32l0.cpu -- clearing lockup after double fault
Warn : couldn't use loader, falling back to page memory writes
wrote 114688 bytes from file .pioenvs/my_device/firmware.elf in 13.385812s (8.367 KiB/s)
** Programming Finished **
** Verify Started **
verified 112048 bytes in 0.584776s (187.118 KiB/s)
** Verified OK **
** Resetting Target **
adapter speed: 300 kHz
shutdown command invoked
```

Same issue happens with ST-Link in Nucleo boards as uploader.

Is it possible to provide update for OpenOCD?

ivankravets, 2019-02-21:
We've updated openOCD today. Could you try an upstream version of dev/platform? See https://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions

Does it work?

martinjaeger, 2019-02-22:
Awesome, it works perfectly. Flashed several times and I also don't get random flashing issues anymore.

Thanks a lot, Ivan!

closed 2019-02-22 by martinjaeger

martinjaeger, 2019-03-31:
I realized that it still doesn't work with ST-Link (instead of J-Link as used above). The problem seems to be the `WORKAREASIZE` setting in OpenOCD. If it is reduced to `0x1000` in `~/.platformio/packages/tool-openocd/scripts/board/st_nucleo_l073rz.cfg`, everything works fine.

As this setting should be changed in OpenOCD directly, I opened a ticket [there](https://sourceforge.net/p/openocd/tickets/231/).