platformio/platform-ststm32 #177: STM32F303CBT6 Support?

glynhudson, 2019-01-10:
Hi, 

Would it be possible to add support for STM32F303CBT6? It's very similar to other MCU's supported by pio but with some extra pins:

https://uk.farnell.com/stmicroelectronics/stm32f303cbt6/mcu-32bit-cortex-m4-72mhz-lqfp/dp/2333254

https://datasheet.octopart.com/STM32F303CBT6-STMicroelectronics-datasheet-13725529.pdf

We're using this chip for new OpenEnergyMonitor hardware, it would be really great to be able to continue developing using platformio 

Cc. @trystanlea @danbates2

ivankravets, 2019-01-10:
Hi Glyn,

Which framework do you use? Arduino?

TrystanLea, 2019-01-10:
Hello @ivankravets we are using stm32cube, thanks for your quick reply.

valeros, 2019-01-10:
Hi @glynhudson @TrystanLea ! New `genericSTM32F303CB` board has been added. Would be great if you test it with your project. Just install ststm32 platform from dev branch.

ivankravets, 2019-01-10:
See https://docs.platformio.org/en/latest/platforms/ststm32.html#upstream

TrystanLea, 2019-01-10:
Thanks @valeros @ivankravets really appreciated

glynhudson, 2019-01-10:
Wow, that was quick! Thanks a lot of guys. We'll try to test as soon as possible. I'm assuming we will need to install the dev version of Pio?

glynhudson, 2019-01-14:
Woop woop! That works great. Thanks a lot guys :+1: 

I've added a `platfromio.ini` to our project:

https://github.com/openenergymonitor/STM32/commit/96b0176eb424e00c3b7c6e62f95b6508e65c4cde

closed 2019-01-14 by glynhudson

Zdenko-Stanec, 2024-01-07:
Hi, 

Is it possible to add support for the Arduino framework for this MCU?

Thank you,