platformio/platform-ststm32 #178: SerialUSB on bluepill_f103c8/genericSTM32F103C8

LexNetAT, 2019-01-11:
Hi,

I need help or something like that. Currently I can only use an older version and cannot update to a new version (like 4.6.0) because otherwise there is no SerialUSB anymore:

This is working:
>[env:bluepill_f103c8]
>platform = ststm32@<4.0.0
>board = bluepill_f103c8
>framework = arduino

This doesn't work:
>[env:bluepill_f103c8]
>platform = ststm32@~4.6.0
>board = bluepill_f103c8
>framework = arduino

If tested with ArduinoIDE and stm32duino@2018.12.3 all functions seems to work including the DTR-Problem. Unfortunately, the DTR issue is still present in the last version (which is 3.8.0).

Thank you in advance!

xfdr0805, 2019-01-15:
I found the core changed from Maple based core to STM's HAL based core when updated 4.6.0
many old codes can not build I dont know why?

LexNetAT, 2019-01-17:
The question is, why not which is a basic function of the STM32.
To my knowledge, other platforms from STM32 support HID devices and much more, like MIDI and storage.

ivankravets, 2019-02-21:
See https://github.com/platformio/platform-ststm32/issues/169

closed 2019-02-21 by ivankravets

chechu2k5, 2019-02-26:
For what it is worth, I got USB Serial working with the latest HAL based arduino core.

I found out enabling USB Serial requires the following defines based on the contents of platform.txt in the platform-ststm32 folder

```
# USB Flags
# ---------
build.usb_flags=-DUSBCON {build.usb_speed} -DUSBD_VID={build.vid} '-DUSB_MANUFACTURER={build.usb_manufacturer}' '-DUSB_PRODUCT="{build.board}"' -DHAL_PCD_MODULE_ENABLED 
```

So, I figured passing the required definitions via build_flags in platformio.ini should work

Platformio.ini

```ini
[env:bluepill_f103c8]
platform = https://github.com/platformio/platform-ststm32
board = bluepill_f103c8
framework = arduino
build_flags = -D USBCON 
              -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
              -D HAL_PCD_MODULE_ENABLED
              -D USBD_VID=0x1EAF
              -D USBD_PID=0x0004
              -D USB_PRODUCT=\"bluepill\"
```

main.cpp

```c++
#include <Arduino.h>
#include <USBSerial.h>

void setup() {
  SerialUSB.begin();
}

void loop() {
}
```

* You will need to correct the vid, pid and product names
* You need to escape the double quote for the string with a backslash, otherwise you will encounter compilation errors

HTH