platformio/platform-ststm32 #179: Request: STM32F1xx SPL support

Johanneslueke, 2019-01-13:
I have recently started platform IO but i am missing support for STM32F107VC, would you mind adding CMSIS / SPL support for this platform?

shineyruan, 2019-02-22:
Same. I'm waiting for STM32F1xx SPL support too. It would be a great help for me if added.

pr1metine, 2022-11-30:
Our uni class requires us to use this. I'll see what I can do.

pr1metine, 2022-11-30:
So I came up with a super hacky but working solution. 

- Additional source files needed to be [added to framework-spl](https://github.com/pr1metine/platformio-pkg-framework-spl/commit/6bc4be1bb9466df28f8d44295339c4d11472fb33)
- I had [to remove the `-nostartfiles` link flag after which I added spl to `nucleo_f103rb.json` ](https://github.com/pr1metine/platform-ststm32/commit/88f5982f67bee7307fb1f1515551bcc94f531f36)

Next time, I want to investigate the linking issues and iron out further problems.