platformio/platform-ststm32 #181: BOARD_NAME define conflicts with Marlin

ktand, 2019-01-14:
In the latest release there is a `BOARD_NAME` defined in the GCC command line. Unfortunately this conflicts with the `BOARD_NAME` definition in the Marlin project (https://github.com/MarlinFirmware/Marlin)

**stm32duino.py**

```
    CPPDEFINES=[
        series,
        "HAL_UART_MODULE_ENABLED",
        ("ARDUINO", 10805),
        ("ARDUINO_%s" % variant),
        ("ARDUINO_ARCH_STM32"),
        ("BOARD_NAME", variant),
        env.BoardConfig().get("build.variant", "").upper()
    ],
```

I think it was introduced when support for Arduino Core STM32 was added (https://github.com/platformio/platform-ststm32/pull/151)

What is the usage for the `BOARD_NAME` define? Is it needed or could it be removed or renamed?

Regards,
Karl

fpistm, 2019-01-14:
Hi @ktand ,  this is not linked to pio. This is one defined by the core:
https://github.com/stm32duino/Arduino_Core_STM32/commit/ca62452122d7da8ac08e2b2212fc899575fbe31d
It was requested by several users to know the board name.

ktand, 2019-01-14:
> stm32duino/Arduino_Core_STM32@ca62452

Hi @fpistm

Ok, I understand. However it seems like BOARD_NAME is defined like:

-DEXAMPLEBOARD -DBOARD_NAME=EXAMPLEBOARD

I thiink there should at least be quotes around -DBOARD_NAME, like:

-DEXAMPLEBOARD -DBOARD_NAME="EXAMPLEBOARD"

Regards,
Karl

fpistm, 2019-01-14:
Right it's a string.

ktand, 2019-01-14:
When I compile with PlatformIO it's defined as:

-DARDUINO_ARMED_V1 -DARDUINO_ARCH_STM32 -DBOARD_NAME=ARMED_V1 -DARMED_V1

Notice the missing quotation marks.

Karl

ivankravets, 2019-02-21:
Just use  https://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-unflags
```
[env:myenv]
board = ...
build_unflags = -DBOARD_NAME=ARMED_V1
```

Did it help?

ktand, 2019-02-21:
@ivankravets 
@fpistm

Yes, sure it will help but why isn't BOARD_NAME defined as a string?

An example, for the board variant "ARMED_V1" the compiler will be invoked with:

-DARDUINO_ARMED_V1 -DARDUINO_ARCH_STM32 -DBOARD_NAME=ARMED_V1 -DARMED_V1

So BOARD_NAME will be defined as ARMED_V1 which is I assume will be defined as 1.

So BOARD_NAME will be equal to 1, or is my CPP knowledge fading away...? :-)

Sure you can do: 

#ifdef BOARD_NAME but I don't think that was the idea with the BOARD_NAME define as you could have used #ifdef ARMED_V1 instead.

ivankravets, 2019-02-21:
@fpistm can we remove `BOARD_NAME` macro? Do you depend on it? This is actually something new to Arduino cores.

fpistm, 2019-02-22:
Don't know how PIO handle platform.txt but with Arduino IDE this is a string.
-DARDUINO_ARMED_V1 -DARDUINO_ARCH_STM32 "-DBOARD_NAME=\"ARMED_V1\"" -DSTM32F407xx

So, I guess this is an issue how PIO parse it as this is not the same behaviour.

ivankravets, 2019-02-22:
Sorry, it should be fixed in the upstream version of dev/platform. Please re-test. See instruction => https://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions

closed 2019-02-22 by ivankravets

ktand, 2019-02-23:
I can confirm that this is fixed in the upstream version - thanks!