platformio/platform-ststm32 #183: Blue Pill linked incorrectly for DFU Bootloader

braiden, 2019-01-22:
Hi,

I'm trying to build a simple blink sketch for blue pill (STM32F103C) and upload over dfu:

```
[env:broken]
platform = ststm32
framework = arduino
board = bluepill_f103c8
upload_protocol = dfu
upload_port = not_used
```

The resulting binary doesn't work. It expects to file `.text` at `0x0800010c`, but with DFU it will be programmed at `0x0802010c`. After some experimentation, overriding `build.variant` works:

```
[env:bluepill]
platform = ststm32
framework = arduino
board = bluepill_f103c8
board_build.variant = stm32f1
upload_protocol = dfu
upload_port = not_used
```

There's code in `builder/frameworks/arduino/maple/stm32f1.py` that correctly links the code, but that doesn't run by default(?). By changing `build.variant` something not in core variants `stm32f1.py` runs.

Seems like a bug? Or maybe i'm missing something.

braiden, 2019-01-24:
A quick update, as I'm still new to stm32 and arduino. You probably know already, there are two cores:

new: https://github.com/stm32duino/Arduino_Core_STM32
old: https://github.com/rogerclarkmelbourne/Arduino_STM32

The old core supports the DFU (rogerclarkmelbourne/STM32duino-bootloader) bootloader I'm using. The newer core does not. Switching variants had the side effect of switching cores which is why my code started running.

As for this bug:

Is there a compelling reason to include DFU as a supported `upload_protocol` for boards which use stm32duino?

When the bootloader exists it occupies flash at 0x0800000 and user sketch must be linked to run at 0x0802000. Since dfu stm32duino doesn't exist in newer stm32duino core, it simply links the code to 0x0800000 and it will always fault when dfu is used.

Listing DFU as a supported `upload_protocol` feels like it will trap people like it did me. 

I see a pull request to add a HID bootloader (https://github.com/Serasidis/STM32_HID_Bootloader) to stm32duino (https://github.com/stm32duino/Arduino_Core_STM32/pull/415). Consider removing DFU and offering this as an alternative?

BennehBoy, 2019-01-24:
@braiden, I've just updated that PR to include DFU support.

ivankravets, 2019-02-21:
See https://github.com/platformio/platform-ststm32/issues/169#issuecomment-466175809

Please re-test.

closed 2019-02-21 by ivankravets