platformio/platform-ststm32 #187: Blue Pill : Tone (arduino-framework) no longer works in STSTM32 4.0.6

LieBtrau, 2019-01-26:
This functionality worked fine in 4.0.5.  Upgrading to 4.0.6 broke it.  
I noticed that 4.0.6 uses the CMSIS libraries, while 4.0.5 relies on a libmaple derived library.
```
void setup() {
  byte pin = PA8;
  tone(pin, 2000, 1000);
}

void loop() {
  // no need to repeat the melody.
}
```
To my mind, it's still too early to migrate the BluePill to CMSIS.  Libmaple derived libraries still offer far way better support.

fpistm, 2019-01-28:
Just FYI,
I've fixed this issue:
https://github.com/stm32duino/Arduino_Core_STM32/commit/7b33ad05680a3b0e6be6cdfe074f0f9ae696055c

ivankravets, 2019-01-31:
Hi @fpistm, when do you plan to have a next release https://github.com/stm32duino/Arduino_Core_STM32/releases?

fpistm, 2019-01-31:
@ivankravets  Soon. Hope in 1 or 2 weeks. Max end of february.

ivankravets, 2019-01-31:
Great! Please ping us here. Thanks!

fpistm, 2019-02-13:
@ivankravets 
1.5.0 is released.

LieBtrau, 2019-02-16:
Unfortunately, the issue isn't fixed.
My main.cpp looks like this:
```
#include <Arduino.h>

void setup() {
  byte pin = PA8;
  tone(pin, 2000);
}

void loop() {
  // no need to repeat the melody.
}
```
My working platform.ini file is this:
```
[env:bluepill_f103c8]
platform = ststm32@4.5.0
board = bluepill_f103c8
framework = arduino
```
When I leave out the @4.5.0 in order to use the installed ST STM32 5.0.0, building and uploading works fine, but there's no output waveform on pin A8.

fpistm, 2019-02-18:
@LieBtrau I guess you need to wait 1.5.0 will be integrated to PIO to test.

ivankravets, 2019-02-21:
See comment https://github.com/platformio/platform-ststm32/issues/169#issuecomment-466175809

closed 2019-02-21 by ivankravets