platformio/platform-ststm32 #191: blue pill dfu upload

exothink, 2019-02-04:
By default PlatformIO selects ‘Arduino_Core_STM32’, it does not support dfu(usb). Adding ‘board_build.variant = BLUEPILL_F103XX’ to ‘platformio.ini’ causes PlatformIO to use the other core ‘Arduino_STM32’. This one does support dfu and everthing works fine.

PlatformIO should not give you the option of using ‘dfu’ for uploading with the blue pill and the ‘Arduino_Core_STM32’ which it defaults to, since it can’t possibly work.

me21, 2019-02-05:
I agree, only I hit this issue while using Mbed. It currently doesn't support dfu either. The best resolution would be to add support of dfu to these frameworks; I think it boils down to adding alternative linker script (see how it's done in rogerclark's Arduino core in PlatformIO).
But at least PlatformIO should warn about using incompatible settings during build.

me21, 2019-02-05:
I forgot about interrupt controller configuration. It should use interrupts table at another address with dfu bootloader. So it can't work with linker script alone, it requires editing Arduino core sources.

ilya89vet, 2019-02-10:
The update 5.0.0 did not fix i :-(

ilya89vet, 2019-02-11:
I try to stitch blinking, is written to consoles that the firmware is successful, BUT the LED does not blink and in a device manager instead of Maple Serial Comport Maple DFU (Bootloader crashes) it appears it is necessary to recover!
My Platformio.ini file
`[env:bluepill_f103c8]
platform = ststm32
board = bluepill_f103c8
framework = arduino
upload_protocol = dfu`

ivankravets, 2019-02-21:
See https://github.com/platformio/platform-ststm32/issues/169

closed 2019-02-21 by ivankravets

me21, 2019-02-21:
I think upstream official Arduino core now supports DFU. At least some work is going on.