platformio/platform-ststm32 #193: Upgrading to 5.0.0 breaks bluepill w/ libopencm3

dramaturg, 2019-02-08:
After upgrading to 5.0.0 building for a bluepill using libopencm3 fails when linking:

```
[Fri Feb  8 22:00:05 2019] Processing bluepill_f103c8 (framework: libopencm3; platform: ststm32; board: bluepill_f103c8)
---------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PLATFORM: ST STM32 > BluePill F103C8
SYSTEM: STM32F103C8T6 72MHz 20KB RAM (64KB Flash)
DEBUG: CURRENT(stlink) EXTERNAL(blackmagic, jlink, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 2 compatible libraries
Scanning dependencies...
No dependencies

[.. truncated "Compiling" messages ...]

Archiving .pioenvs/bluepill_f103c8/libFrameworkLibOpenCM3.a
Linking .pioenvs/bluepill_f103c8/firmware.elf
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\177' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\001' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\001' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\001' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\000' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\000' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\000' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\000' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\000' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\000' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\000' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\000' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\000' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\001' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: ignoring invalid character `\000' in expression
/home/seb/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/bluepill_f103c8/src/main.o:1: syntax error
collect2: error: ld returned 1 exit status
*** [.pioenvs/bluepill_f103c8/firmware.elf] Error 1
================================================================== [ERROR] Took 2.02 seconds ==================================================================
```

Enforcing 4.6.0 like this and we are back to a working state:
```
[env:bluepill_f103c8]
platform = ststm32@~4.6.0
board = bluepill_f103c8
framework = libopencm3

upload_protocol = stlink
```

ivankravets, 2019-02-09:
@valeros please help

valeros, 2019-03-04:
Hi @dramaturg ! Could you please try to compile your project using the latest platform from the dev branch?
```
[env:bluepill_f103c8]
platform = https://github.com/platformio/platform-ststm32.git
board = bluepill_f103c8
framework = libopencm3
```

dramaturg, 2019-03-04:
Same issue, I'm afraid. Here is a "blinky" snippet that shows the issue:
https://github.com/dramaturg/pio-stm32-broken

valeros, 2019-03-04:
Sorry, couldn't reproduce the issue either on Windows/Ubuntu Linux. Please build your project in verbose mode and attach here the log. Thanks!

dramaturg, 2019-03-04:
I found upgrading platformio to the latest release fixes things for me. I saw the issue with 3.5.3 whereas 3.6.4 compiles fine.

valeros, 2019-05-09:
Please reopen the issue if you still have the same problem.

closed 2019-05-09 by valeros