platformio/platform-ststm32 #195: not created a project for bluepill_f103c8_128k

nikantovik, 2019-02-10:
PIO Core Call Error: “The next files/directories have been created in /home/asus/Documents/PlatformIO/Projects/name\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\nError: Processing bluepill_f103c8_128k (platform: ststm32; board: bluepill_f103c8_128k; framework: mbed)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via -v, --verbose option\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8_128k.html\nPLATFORM: 1 ST STM32 > BluePill F103C8 (128k)\nHARDWARE: STM32F103C8T6 72MHz 20KB RAM (128KB Flash)\nDEBUG: CURRENT(stlink) EXTERNAL(blackmagic, jlink, stlink)\nwrite() takes exactly 2 arguments (3 given)\nmbed build API internal error\n========================== [ERROR] Took 0.71 seconds ==========================”

krazatchu, 2019-02-11:
Same problem here, tried reinstalling, deleted the .platformio folder, etc. 
It can't create a new project with the bluepill 128k. 
Also tried changing the board type in the INI to the 128k and got the same error: 
           
         write() takes exactly 2 arguments (3 given)

valeros, 2019-02-12:
Hi @nikantovik @krazatchu ! `mbed` framework was erroneously added to the framework list for this board. `mbed` supports only regular `bluepill_f103c8` with 64K flash. 
I'd recommend the next workaround for your case:
Create your project for `bluepill_f103c8` but in `build_flags` field add a custom ld-script with altered flash size:
```
[env:bluepill_f103c8]
platform = ststm32
framework = mbed
board = bluepill_f103c8
build_flags = -Wl,-T"mbed_bluepull_128.ld"
```
This [mbed_bluepull_128.ld](https://pastebin.com/MpUPRXf5) you need to copy to this folder:
`/home/YOUR_USER/.platformio/platforms/ststm32/ldscripts` for Linux
`c:\Users\YOUR_USER\.platformio\platforms\ststm32\ldscripts` for Windows

me21, 2019-02-12:
How come `mbed` can support only Blue Pill 64K when the only difference is more flash memory?

nikantovik, 2019-02-13:
> Hi @nikantovik @krazatchu ! `mbed` framework was erroneously added to the framework list for this board. `mbed` supports only regular `bluepill_f103c8` with 64K flash.
> I'd recommend the next [...]

[env:genericSTM32F103C8]
platform = ststm32@4.6.0
board = genericSTM32F103C8
framework = mbed
board_upload.maximum_size = 120000 ; 

it's okay

krazatchu, 2019-02-13:
@nikantovik, I tried your INI and it works for me. 
@valeros, your suggestion works as well, it does throw a warning but I can ignore it.

krazatchu, 2019-02-13:
This works as well, but does result in much larger (+10k) compile for my project. 
I can't seem to combine the build flags with -D MBED_BUILD_PROFILE_RELEASE.

[env:genericSTM32F103C8]
platform = ststm32 ;@4.6.0
board = genericSTM32F103C8
framework = mbed
board_upload.maximum_size = 120000
build_flags = -Wl,-T"mbed_bluepull_128.ld"

valeros, 2019-02-13:
@me21 Mainly because there is only one [linker script](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F1/TARGET_BLUEPILL_F103C8/device/TOOLCHAIN_GCC_ARM/STM32F103XB.ld) for 64KB version.

valeros, 2019-02-13:
@krazatchu  As for the memory footprint, you can try to use the nano version of the standard library. You'll need to create an extra script (e.g. `nano.py`) in the root folder of your project with the next content:
```
Import("env")
env.Append(LINKFLAGS=["--specs=nano.specs"])
```
And then enable it in your `platformio.ini`
```
[env:genericSTM32F103C8]
platform = ststm32 ;@4.6.0
board = genericSTM32F103C8
framework = mbed
board_upload.maximum_size = 120000
build_flags = -Wl,-T"mbed_bluepull_128.ld
extra_scripts = nano.py
```

@nikantovik Your config is fine with one important coveat: the resulting firmware will be linked using ld-script for 64KB version.