platformio/platform-ststm32 #199: version 5.0, using mbed framework the C++ Tips Failure

chengxg, 2019-02-19:
The c++ prompt is invalid as long as "# include < mbed.h>" is referenced.
[image]

vscode output errors:
[Error - 上午9:52:10] Connection to server got closed. Server will not be restarted.
[Error - 上午9:52:10] Request textDocument/definition failed.
Error: Connection got disposed.
	at Object.dispose (C:\Users\24960\.vscode\extensions\ms-vscode.cpptools-0.21.0\node_modules\vscode-jsonrpc\lib\main.js:876:25) ...

ivankravets, 2019-02-21:
It seems like VSCode issue. How to reproduce it?

chengxg, 2019-02-22:
> It seems like VSCode issue. How to reproduce it?

I don't know,but I tried v4.6 without this problem.

ivankravets, 2019-02-22:
@valeros Please help.

ivankravets, 2019-03-03:
Fixed in the latest release. Please open PIO IDE Terminal and type `pio update`. Restart IDE.

closed 2019-03-03 by ivankravets