platformio/platform-ststm32 #201: STM32F103C8 BluePill: Error: invalid ELF file, no program headers

JarrettR, 2019-02-21:
Nearly default blink project targeting a Blue Pill board with an external STLink. 

Tested on 5.0.0 and also latest 5.1.0, not tested on anything earlier.

ini file:
```
[env:bluepill_f103c8]
platform = ststm32
framework = libopencm3
board = bluepill_f103c8
debug_tool = stlink
upload_protocol = stlink
build_flags = -Wl,-Map,output.map
```

Output is:

```
Processing bluepill_f103c8 (platform: ststm32; board: bluepill_f103c8; framework: libopencm3)
---------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
PLATFORM: ST STM32 > BluePill F103C8
HARDWARE: STM32F103C8T6 72MHz 20KB RAM (64KB Flash)
DEBUG: CURRENT(stlink) EXTERNAL(blackmagic, jlink, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 0 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs\bluepill_f103c8\src\main.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\f1\adc.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\adc_common_v1.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\can.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\desig.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\f1\ethernet.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\f1\flash.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\f1\gpio.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\f1\rcc.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\f1\rtc.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\f1\timer.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\ethernet\mac.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\ethernet\mac_stm32fxx7.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\ethernet\phy.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\ethernet\phy_ksz8051mll.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\crc_common_all.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\dac_common_all.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\dma_common_l1f013.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\gpio_common_all.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\i2c_common_all.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\iwdg_common_all.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\pwr_common_all.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\spi_common_all.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\spi_common_l1f124.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\timer_common_all.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\usart_common_all.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\usart_common_f124.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\rcc_common_all.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\exti_common_all.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\flash_common_f01.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\usb\usb.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\usb\usb_control.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\usb\usb_standard.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\usb\usb_msc.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\usb\usb_fx07_common.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\usb\usb_f107.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\common\st_usbfs_core.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\stm32\st_usbfs_v1.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\cm3\vector.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\cm3\systick.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\cm3\scb.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\cm3\nvic.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\cm3\assert.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\cm3\sync.o
Compiling .pioenvs\bluepill_f103c8\FrameworkLibOpenCM3\lib\cm3\dwt.o
Archiving .pioenvs\bluepill_f103c8\libFrameworkLibOpenCM3.a
Linking .pioenvs\bluepill_f103c8\firmware.elf
c:/users/jrain/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: warning: cannot find entry symbol _start; defaulting to 00008000
Checking size .pioenvs\bluepill_f103c8\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   0.0% (used 0 bytes from 20480 bytes)
PROGRAM: [          ]   0.0% (used 0 bytes from 65536 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, dfu, jlink, mbed, stlink
CURRENT: upload_protocol = stlink
Uploading .pioenvs\bluepill_f103c8\firmware.elf
GNU MCU Eclipse 64-bits Open On-Chip Debugger 0.10.0+dev-00392-gbe9ef0b0 (2018-01-12-15:03)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
hla_swd
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 1000 kHz
adapter_nsrst_delay: 100
none separate
none separate
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : Unable to match requested speed 1000 kHz, using 950 kHz
Info : clock speed 950 kHz
Info : STLINK v2 JTAG v17 API v2 SWIM v4 VID 0x0483 PID 0x3748
Info : using stlink api v2
Info : Target voltage: 3.287408
Info : stm32f1x.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x08005770 msp: 0x20005000
** Programming Started **
auto erase enabled
Error: invalid ELF file, no program headers
embedded:startup.tcl:476: Error: ** Programming Failed **
in procedure 'program'
in procedure 'program_error' called at file "embedded:startup.tcl", line 532
at file "embedded:startup.tcl", line 476
*** [upload] Error 1
============================================================== [ERROR] Took 13.79 seconds ==============================================================
The terminal process terminated with exit code: 1
```

I'd suspect that it's a linker issue, with data/program memory showing 0% utilisation, and the 0x8000 start not being defined, which I'd expect it to. The ELF file is created, and is about 600 bytes.

JarrettR, 2019-02-21:
To narrow down troubleshooting:
These are the changes I made:
https://github.com/platformio/platform-ststm32/compare/develop...JarrettR:develop
(As an aside, if this issue gets resolved, could this get merged in?)

And the default CMSIS example works properly, so it's not my programmer environment.

valeros, 2019-03-04:
Hi @JarrettR ! Could you please try to compile your project using the latest platform from the dev branch? 
```
[env:bluepill_f103c8]
platform = https://github.com/platformio/platform-ststm32.git
framework = libopencm3
board = bluepill_f103c8
debug_tool = stlink
upload_protocol = stlink
build_flags = -Wl,-Map,output.map
```

JarrettR, 2019-03-05:
Compile output is now:
```Linking .pioenvs\bluepill_f103c8\firmware.elf
Building .pioenvs\bluepill_f103c8\firmware.bin
Checking size .pioenvs\bluepill_f103c8\firmware.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [          ]   0.0% (used 0 bytes from 20480 bytes)
PROGRAM: [          ]   1.0% (used 676 bytes from 65536 bytes)```

Which looks more like what I'd expect! 

I'm not around my hardware for a couple days to test properly, though. Will test and update when I can.

JarrettR, 2019-04-09:
Sorry about the delay, had a really good excuse.

I just tested this on 5.2.0, and the latest release (5.3.0), and it works great on both, thanks!

closed 2019-04-09 by JarrettR

mprymek, 2019-05-23:
FYI, the same problem still persists for
```
[env:bluepill_f103c8_128k]
platform = ststm32
# Does not solve anything
#platform = https://github.com/platformio/platform-ststm32.git
board = bluepill_f103c8_128k
framework = libopencm3
```