platformio/platform-ststm32 #202: Problem on using new mbed_app.json config file

masoudr, 2019-02-22:
Hi,
I'm trying to use new `mbed_app.json` configuration based on [this](https://docs.platformio.org/en/latest/frameworks/mbed.html#the-configuration-system), but I have some problem and I think it is a PIO related issue.
Before, I used this configuration which was work fine:
```
[env:disco_f407vg]
platform = ststm32
board = disco_f407vg
framework = mbed
upload_protocol = stlink
build_flags = -D MBED_CONF_TARGET_STDIO_UART_RX=PA_10 -D MBED_CONF_TARGET_STDIO_UART_TX=PA_9
     -D PIO_FRAMEWORK_MBED_RTOS_PRESENT -D PIO_FRAMEWORK_MBED_EVENTS_PRESENT 
     -D MBED_CONF_TARGET_LSE_AVAILABLE=1

monitor_speed = 115200
```
Now I created a new `mbed_app.json` and removed the build flags from the old config.
But I have many problems on founding headers like:
`SDBlockdevice.h` not found
`mbedtls/base64.h` not found (other mbedtls headers also not found)
I tried to add configs as standard configuration but I get the same error.
For example, if I want to enable the `SD` I tried this:
```
{
    "macros": [
        "MBEDTLS_USER_CONFIG_FILE=\"mbedtls_config.h\""
    ],
    "config": {
    },
    "target_overrides": {
        "*": {
            "target.features_add": ["STORAGE"],
            "target.components_add": ["SD"],
            "platform.stdio-baud-rate": 115200,
            "target.stdio_uart_tx": "PA_9",
            "target.stdio_uart_rx": "PA_10"
        }
    }
}
```
Which seems fine based on [ARM docs](https://os.mbed.com/docs/mbed-os/v5.11/reference/storage.html#Default-BlockDevice-configuration).
I also checked the `mbed_config.h` file and there was no definition for `SDBlockdevice`.

valeros, 2019-02-22:
Hi @masoudr ! Could you please share a minimal example to reproduce the issue? Without the specific macro `mbedtls_config.h` it should work with this combination:
```json
{
    "target_overrides": {
        "*": {
            "target.features_add": ["STORAGE"],
            "target.components_add": ["SD"],
            "platform.stdio-baud-rate": 115200,
            "target.stdio_uart_tx": "PA_9",
            "target.stdio_uart_rx": "PA_10"
        }
    }
}
```
```ini
[env:disco_f407vg]
platform = ststm32
framework = mbed
board = disco_f407vg
build_flags = -D PIO_FRAMEWORK_MBED_RTOS_PRESENT 
```

masoudr, 2019-02-23:
Hi,
Thanks, it seems that I should place `build_flags = -D PIO_FRAMEWORK_MBED_RTOS_PRESENT` in the PIO config file. Now it is working.

closed 2019-02-23 by masoudr