platformio/platform-ststm32 #203: Home: Could not initialize project

zhaojinlong, 2019-02-23:
PIO Core Call Error: "The next files/directories have been created in E:\\project\\stm32\\test_stm32\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\r\nPlatformManager: Installing ststm32\r\nDownloading\r\nDownloading\r\nUnpacking\r\nUnpacking\r\nststm32 @ 5.0.0 has been successfully installed!\r\nPackageManager: Installing toolchain-gccarmnoneeabi @ >=1.60301.0\r\r\nDownloading\r\nDownloading\r\nUnpacking\r\nUnpacking\r\ntoolchain-gccarmnoneeabi @ 1.70201.0 has been successfully installed!\r\nThe platform 'ststm32' has been successfully installed!\r\r\nThe rest of packages will be installed automatically depending on your build environment.\n\nError: Processing genericSTM32F103C8 (platform: ststm32; board: genericSTM32F103C8; framework: mbed)\r\n\r\nPackageManager: Installing framework-mbed @ ~5.51103.190202\r\nDownloading\r\nDownloading\r\nDownloading...\r\nUnpacking\r\nUnpacking\r\nVerbose mode can be enabled via `-v, --verbose` option\r\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103C8.html\r\nPLATFORM: ST STM32 > STM32F103C8 (20k RAM. 64k Flash)\r\nHARDWARE: STM32F103C8T6 72MHz 20KB RAM (64KB Flash)\r\nDEBUG: CURRENT(blackmagic) EXTERNAL(blackmagic, jlink, stlink)\r\nmbed build API internal error\r\npath is on drive C:, start on drive E:\r\n [ERROR] Took 1323.95 seconds"

ivankravets, 2019-02-23:
Could you reproduce this issue with the latest version? Please run in PIO IDE terminal `pio update` and  try buidling again.

valeros, 2019-05-09:
Feel free to reopen the issue if you still have the same problem.

closed 2019-05-09 by valeros