platformio/platform-ststm32 #204: Home: Could not initialize project with mbed framework python 3.7 fedora 29 x86_64

andrewklmn, 2019-02-23:
ImportError: No module named lib2to3.pgen2.parse
=========================================
python version is 3.7 on fedora 29 x86_64
python2-devel is installed 
python3-devel is installed 
files here:
/usr/lib/python3.7/lib2to3
/usr/lib64/python2.7/lib2to3
/usr/lib64/python3.7/lib2to3
/usr/bin/2to3
Arduino framework is ok. And stm32cube - too...

PlatformIO error message:
=======================
PIO Core Call Error: "The next files/directories have been created in /home/user/Documents/PlatformIO/Projects/mbed-test\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\nError: Processing nucleo_f103rb (platform: ststm32; board: nucleo_f103rb; framework: mbed)\n--------------------------------------------------------------------------------------------------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/ststm32/nucleo_f103rb.html\nPLATFORM: ST STM32 > ST Nucleo F103RB\nHARDWARE: STM32F103RBT6 72MHz 20KB RAM (128KB Flash)\nDEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)\nImportError: No module named lib2to3.pgen2.parse:\nFile \"/home/user/.platformio/penv/lib/python2.7/site-packages/platformio/builder/main.py\", line 173:\nenv.SConscript(\"$BUILD_SCRIPT\")\nFile \"/home/user/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 541:\nreturn _SConscript(self.fs, *files, **subst_kw)\nFile \"/home/user/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 250:\nexec _file_ in call_stack[-1].globals\nFile \"/home/user/.platformio/platforms/ststm32/builder/main.py\", line 90:\ntarget_elf = env.BuildProgram()\nFile \"/home/user/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py\", line 224:\nreturn self.method(*nargs, **kwargs)\nFile \"/home/user/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py\", line 107:\nenv.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))\nFile \"/home/user/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py\", line 224:\nreturn self.method(*nargs, **kwargs)\nFile \"/home/user/.platformio/penv/lib/python2.7/site-packages/platformio/builder/tools/platformio.py\", line 319:\nSConscript(env.GetFrameworkScript(f), exports=\"env\")\nFile \"/home/user/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 604:\nreturn method(*args, **kw)\nFile \"/home/user/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 541:\nreturn _SConscript(self.fs, *files, **subst_kw)\nFile \"/home/user/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 250:\nexec _file_ in call_stack[-1].globals\nFile \"/home/user/.platformio/platforms/ststm32/builder/frameworks/mbed.py\", line 37:\n\"platformio-build.py\"))\nFile \"/home/user/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 604:\nreturn method(*args, **kw)\nFile \"/home/user/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 541:\nreturn _SConscript(self.fs, *files, **subst_kw)\nFile \"/home/user/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 250:\nexec _file_ in call_stack[-1].globals\nFile \"/home/user/.platformio/packages/framework-mbed/platformio/platformio-build.py\", line 39:\nfrom pio_mbed_adapter import PlatformioMbedAdapter\nFile \"/home/user/.platformio/packages/framework-mbed/platformio/pio_mbed_adapter.py\", line 23:\nfrom tools.build_api import (prepare_toolchain, merge_region_list,\nFile \"/home/user/.platformio/packages/framework-mbed/tools/build_api.py\", line 48:\nfrom .toolchains import TOOLCHAIN_CLASSES\nFile \"/home/user/.platformio/packages/framework-mbed/tools/toolchains/__init__.py\", line 40:\nfrom ..notifier.term import TerminalNotifier\nFile \"/home/user/.platformio/packages/framework-mbed/tools/notifier/term.py\", line 17:\nfrom past.builtins import basestring\nFile \"/home/user/.platformio/packages/framework-mbed/platformio/package_deps/py2/past/__init__.py\", line 88:\nfrom past.translation import install_hooks as autotranslate\nFile \"/home/user/.platformio/packages/framework-mbed/platformio/package_deps/py2/past/translation/__init__.py\", line 41:\nfrom lib2to3.pgen2.parse import ParseError\n======================================================================= [ERROR] Took 1.14 seconds ======================================================================="

ivankravets, 2019-02-23:
Do you use PlatformIO 4.0 dev?

andrewklmn, 2019-02-24:
I use PlatformIO IDE -Atom - Home 2.0.0·Core 4.0.0a3
After yesterday core and libraries update my project based on mbed framework doesn't compile anymore... It's sad...

ivankravets, 2019-02-24:
Do you have Python 2.7 in a system?

andrewklmn, 2019-02-25:
I set it manualy:
[user@localhost ~]$ alternatives --list | grep python
python	manual	/usr/bin/python2.7

But I still have the same error when I try to add new project based on mbed framework....
========================
ImportError: No module named lib2to3.pgen2.parse

ivankravets, 2019-07-18:
Could you reproduce this issue with the latest PIO Core 4.0? It supports Python3 too.