platformio/platform-ststm32 #205: `lib_archive = false` not disabling library archiving

jeffstaley, 2019-02-23:
Linker is finding `weak` symbols instead of the expected normal functions because framework libraries are archived. Placing `lib_archive = false` in my `platformio.ini` does not disable library archiving.

PlatformIO Version:  3.6.4

`platform.ini`:

```
[env:foo]
lib_archive = false
platform = ststm32
board = disco_f746ng
framework = arduino
build_unflags = -Os
build_flags = -g3 -Og -Wl,-Map,$BUILD_DIR/firmware.map  
```

ivankravets, 2019-02-23:
Oh... I don't like this hack. I move this issue to ST STM32 dev/platform. PlatformIO Core is not responsible for framework building. I agree with you, `lib_archive = false` flag should work for framework too. I've just fixed build script for Arduino ST STM32. Please re-run `$ pio update` and re-test with the upstream version of dev/platform => https://docs.platformio.org/en/latest/platforms/ststm32.html#stable-and-upstream-versions

Does it work now?

jeffstaley, 2019-02-23:
> Does it work now?

Mostly.
Yes.  The `lib_archive = false` option works fine now.   Does exactly what I would expect. Thanks.

But the upstream version is creating a link error due to multiple extraneous `#define` in the linker script?!?!.  

If I manually strip those out it all works fine.

Here's the error:
```
/Users/jeffy/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/foo/preproc.ld:1: ignoring invalid character `#' in expression
/Users/jeffy/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld:.pioenvs/foo/preproc.ld:1: syntax error
collect2: error: ld returned 1 exit status
*** [.pioenvs/foo/firmware.elf] Error 1
```

and here's the bad linker script `preproc.ld`:

[preproc.ld.zip](https://github.com/platformio/platform-ststm32/files/2897358/preproc.ld.zip)

ivankravets, 2019-02-23:
Resolved in https://github.com/platformio/platform-ststm32/commit/6c4bc60683381634fc5c49e456390dcbce5be649

Please again `$ pio update`.

jeffstaley, 2019-02-23:
Updated to 6c4bc60, but still getting the same linker error.

ivankravets, 2019-02-23:
@valeros could you help?

valeros, 2019-02-24:
Hi @jeffstaley ! Could you please run `pio update` (pio will update the Arduino framework) again and try to build your project? Linker scripts for some boards in the Arduino framework require a special kind of preprocessing (filling memory size variables). The reason why this preprocessing fails is because of the flag `-g3`. It forces the compiler to generate extra debug information that apparently breaks linker script syntax.

jeffstaley, 2019-02-25:
OK, that did it. All good now. Thanks for all help.

closed 2019-02-25 by jeffstaley

gagipro, 2023-02-25:
Hello,

I wonder if there can be the same issue for : 
platform        = ststm32
framework       = cmsis

thanks in advance,
regards