platformio/platform-ststm32 #206: maple boards inconsistent framework STM32 STM32F1 folder selection

SevenW, 2019-02-25:
when specifying
```
[env:maple]
platform = ststm32
framework = arduino
board = maple
```
core libraries form ```.platformio\packages\framework-arduinoststm32\STM32``` are used. However, when using one of the three other maple boards, for example ```maple_ret6``` the (preferred) STM32F1 folder is selected: ```.platformio\packages\framework-arduinoststm32\STM32F1```.

Trying to work around this issue, I did not find a way to solve it through platformio.ini (unexperienced user!), but found that I could select the stm32duino STM32F1 libmaple libraries by making a change in boards/maple.json in this repositry. When changing this line  https://github.com/platformio/platform-ststm32/blob/f024ab4ed607c1405d67fa5d7e3c0085acc93072/boards/maple.json#L19 into```"variant": "stm32f1"``` (consistent with the other maple boards) it selects the desired libmaple libs.

Please note that the current maple.json actually selects a Maple Mini, which is not correct.

I was unable to figure out the source of the STM32 folder. The stm32duino repositories only contain STM32F1 and STM32F4 folders.

SevenW, 2019-02-25:
To demonstrate the unexpected result, I include c_cpp_properties.json, where I use the instructions about selecting the proper(?) build core like this:
```
[env:maple]
platform = ststm32
board = maple
framework = arduino
board_build.core = maple
```
as described here: https://docs.platformio.org/en/latest/platforms/ststm32.html#configuration
```
{
    "!!! WARNING !!!": "PLEASE DO NOT MODIFY THIS FILE! USE https://docs.platformio.org/page/projectconf/section_env_build.html#build-flags",
    "configurations": [
        {
            "name": "Win32",
            "includePath": [
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/cores/arduino/avr",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/cores/arduino/stm32",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/cores/arduino/stm32/LL",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/Drivers/STM32F1xx_HAL_Driver/Inc",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/Drivers/STM32F1xx_HAL_Driver/Src",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/STM32F1xx",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/variants/MAPLEMINI_F103CB/usb",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/CMSIS/CMSIS/Core/Include",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/cores/arduino",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/variants/MAPLEMINI_F103CB",
                "C:/Users/SevenW/.platformio/lib/SdFat_ID322/src",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/libraries/SPI/src",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/libraries/CapacitiveSensor-0.5.1",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/libraries/EEPROM/src",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/libraries/IWatchdog/src",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/libraries/Keyboard/src",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/libraries/Mouse/src",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/libraries/Servo/src",
                "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/libraries/Wire/src",
                "C:/Users/SevenW/.platformio/packages/tool-unity",
                "d:/SevenW/PlatformIO/Projects/SDCard/include",
                "d:/SevenW/PlatformIO/Projects/SDCard/src",
                ""
            ],
            "browse": {
                "limitSymbolsToIncludedHeaders": true,
                "databaseFilename": "${workspaceRoot}/.vscode/.browse.c_cpp.db",
                "path": [
                    "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/cores/arduino/avr",
                    "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/cores/arduino/stm32",
                    "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/cores/arduino/stm32/LL",
                    "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/Drivers/STM32F1xx_HAL_Driver/Inc",
                    "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/Drivers/STM32F1xx_HAL_Driver/Src",
                    "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/STM32F1xx",
                    "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/variants/MAPLEMINI_F103CB/usb",
                    "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc",
                    "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src",
                    "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Inc",
                    "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/system/Middlewares/ST/STM32_USB_Device_Library/Class/CDC/Src",
                    "C:/Users/SevenW/.platformio/packages/framework-arduinoststm32/STM32/CMSIS/CMSIS/Core/Include",
[... cut ...]
                "MAPLEMINI_F103CB",
                ""
            ],
            "intelliSenseMode": "clang-x64",
            "cStandard": "c11",
            "cppStandard": "c++14",
            "compilerPath": "C:/Users/SevenW/.platformio/packages/toolchain-gccarmnoneeabi/bin/arm-none-eabi-gcc.exe -mcpu=cortex-m3 -mthumb"
        }
    ],
    "version": 4
}
```

This must lead to unexpected mixtures of the STM32F103RB and STM32F103CB boards, given the defines. Include paths are the STM32 folder, not the STM32F1.

SevenW, 2019-02-25:
... and actually further understanding the systematics of board.json and platform.ini, the following results in a proper build of the libmaple-style code:
```[env:maple]
platform = ststm32
board = maple
framework = arduino
;board_build.core = maple
board_build.variant = maple
```
This is different from the suggestion made by the documentation. Although I can now build this without modifying the platform, I still feel that the behaviour is not as expected, and documented.

valeros, 2019-02-25:
Hi @SevenW ! Thanks for the report, but I can't reproduce the issue. Do you use the latest ststm32 platform version?

SevenW, 2019-02-25:
Fresh install of Visual Studio Code, from there Externsion Platform IO (v1.6.0) and form there ST STM32 Platform V5.1.0. All stock. ~~No earlier history. I am new to platformio since  7days.~~
... Not exeactly right. I first installed it into Atom, a week ago and then 4 days back in visual code.

Should I remove everything and try a fresh install?

valeros, 2019-02-25:
Just run `pio update` from the internal terminal.

SevenW, 2019-02-25:
Only a compiler updated.
```
pio update
Updating tool-scons                      @ 2.20501.7      [Up-to-date]
Updating tool-unity                      @ 1.20403.0      [Up-to-date]
Updating contrib-pysite                  @ 2.27.181225    [Up-to-date]
Updating contrib-piohome                 @ 2.0.0          [Up-to-date]
Updating tool-pioplus                    @ 2.1.0          [Up-to-date]

Platform Manager
================
Platform Atmel AVR
--------
Updating atmelavr                        @ 1.12.5         [Up-to-date]
Updating toolchain-atmelavr              @ 1.50400.0      [Up-to-date]
Updating framework-arduinoavr            @ 2.10623.190209 [Up-to-date]

Platform ST STM32
--------
Updating ststm32                         @ 5.1.0          [Up-to-date]
Updating tool-stm32duino                 @ 1.0.1          [Up-to-date]
Updating tool-openocd                    @ 1.1000.6       [Up-to-date]
Updating framework-arduinoststm32        @ 3.10400.181126 [Up-to-date]
Updating toolchain-gccarmnoneeabi        @ 1.70201.0      [1.80201.181220]
Uninstalling toolchain-gccarmnoneeabi @ 1.70201.0:      [OK]
PackageManager: Installing toolchain-gccarmnoneeabi @ 1.80201.181220
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
toolchain-gccarmnoneeabi @ 1.80201.181220 has been successfully installed!

Library Manager
===============
Updating SdFat                           @ 1.0.16         [Up-to-date]
```

Unfortunately, SDFat now no longer builds. Duplicate inline statements. I still can't use board_build.core = maple but need board_build.variant = maple to find files in STM32F1.

valeros, 2019-02-25:
Oh, I see, the new framework package hasn't been released yet. As for now, you can try to install the platform from [the development branch](http://docs.platformio.org/en/latest/platforms/ststm32.html#upstream).

SevenW, 2019-02-25:
With development branch it works. 
```
[env:maple]
platform = https://github.com/platformio/platform-ststm32.git
;platform = ststm32
board = maple
framework = arduino
board_build.core = maple
```

SevenW, 2019-02-25:
Thanks for helping!

closed 2019-02-25 by SevenW