platformio/platform-ststm32 #207: STM32 Blue / black pill dfu upload and usb serial problems

kloppertje, 2019-02-26:
Some time ago I tried to get a black pill board (STM32F103C8T6) running under platformio. I can upload fine via STLinkV2 (clone). However, if I try to send some data over the emulated USB port, it doesn't work. After plugging in the USB cable, I get a Windows eeror message "unknown usb device (device descriptor request failed)". 

Also, when trying to upload using DFU, things go wrong. I can flash the DFU bootloader, and select dfu as upload method in PIO. After flashing the bootloader, a "Maple DFU" device is seen in the category libusb-win32 devices in the device manager on Windows. However, after flashing once from PIO using DFU, the same error as above appears. 

Something seems wrong with the bootloader; I don't know what though.

platformio.ini:
[env:bluepill_f103c8]
platform = ststm32
; platform = https://github.com/platformio/platform-ststm32.git
board = bluepill_f103c8
; board = genericSTM32F103C8
framework = arduino
; upload_protocol = stlink
upload_protocol = dfu
upload_port = anything

main.cpp:
#include <Arduino.h>

#define LED_PIN PB12

void setup() {
  // Serial.begin(115200);
  pinMode(LED_PIN, OUTPUT);
}

void loop() {
  digitalWrite(LED_PIN, 1);
  delay(900);
  digitalWrite(LED_PIN, 0);
  delay(100);
  // Serial.println("abc");
}

valeros, 2019-02-26:
Hi @kloppertje! Thank for the report. Could you please try the next experiment:
1. Delete current platform `pio platform uninstall ststm32` (Please make sure there is no other `ststm32` folders in PlatformIO home dir `C:\Users\YOUR_USER\.platformio\platforms`)
2. Delete stm32 related packages from `C:\Users\YOUR_USER\.platformio\packages` (like framework-arduinoststm32*) and toolchain `toolchain-gccarmnoneeabi`
3. Install the latest platform from the dev branch `pio platform install https://github.com/platformio/platform-ststm32,git`
4. Burn bootloader to your board.
5. Compile and upload firmware in verbose mode (pio run -e genericSTM32F103C8 -v) with this configuration:
```
[env: genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
upload_protocol = dfu
```
6. Describe here results and attach the entire log somewhere.

Thanks in advance! BTW, does everything work in Arduino IDE?

kloppertje, 2019-02-27:
Hi @valeros, thanks for the quick reply! I've executed all of the steps described above. See attached log for the command line output. 

And now it suddenly works! The serial device is visible after uploading, and the serial port outputs data. Is this solely due to the switch from stm32(duino) core to the maple core? The bluepill (and blackpill since one of the latest ststm32 platform versions) definitions use stm32(duino) core, while genericSTM32F103C8 usees maple core, according to the board definitions.

IIRC everything was working fine in Arduino IDE (was some time ago since I last used it in there).
[verboseOutput.txt](https://github.com/platformio/platform-ststm32/files/2911660/verboseOutput.txt)

pfeerick, 2019-03-09:
This is probably another of those "use the `board_build.core = maple` if wanting to use DFU cases", so should be able to be closed now. Going from [this PR](https://github.com/stm32duino/Arduino_Core_STM32/pull/415) we may be able to use the 'official' stm32duino core soon with DFU...

valeros, 2019-03-12:
Please reopen the issue if you still have any problems.

closed 2019-03-12 by valeros

me21, 2019-03-22:
I think installing from git is not necessary now because there's version 5.3.0.

mikegleasonjr, 2020-02-05:
Oh my god, I spent hours trying to figure out why the serial was disappearing after
1. uploading my first sketch
AND
2. resetting the board (or unplugging/re-plugging it)

Thank you so much!

Davincible, 2020-10-14:
@valeros Hi, I am having a similar problem. I can't seem to upload to my STM32F103C8T6 over USB. I tried the steps you suggested, but to no avail.

My platformio.ini:
```
[env:genericSTM32F103C8]
platform = ststm32
board = genericSTM32F103C8
framework = arduino
upload_protocol = dfu
board_build.core = maple
```

If I run `pio run -e genericSTM32F103C8 -v`:
```
rocessing genericSTM32F103C8 (platform: ststm32; board: genericSTM32F103C8; framework: arduino; upload_protocol: dfu; board_build.core: maple)
-------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103C8.html
PLATFORM: ST STM32 (8.1.0) > STM32F103C8 (20k RAM. 64k Flash)
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32-maple 2.10000.200103 (1.0.0) 
 - tool-dfuutil 1.9.200310 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio/build/genericSTM32F103C8/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.9% (used 4280 bytes from 20480 bytes)
Flash: [===       ]  30.6% (used 20064 bytes from 65536 bytes)
.pio/build/genericSTM32F103C8/firmware.elf  :
section            size        addr
.text             14724   134225920
.text.align           4   134240644
.ARM.exidx            8   134240648
.data              3208   536870912
.rodata            2120   134243864
.bss               1072   536874120
.debug_frame       1000           0
.comment            126           0
.ARM.attributes      41           0
Total             22303
======================================================= [SUCCESS] Took 0.43 seconds =======================================================
```

If I run `pio run -t upload -v`
```
Processing genericSTM32F103C8 (platform: ststm32; board: genericSTM32F103C8; framework: arduino; upload_protocol: dfu; board_build.core: maple)
-------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103C8.html
PLATFORM: ST STM32 (8.1.0) > STM32F103C8 (20k RAM. 64k Flash)
HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32-maple 2.10000.200103 (1.0.0) 
 - tool-dfuutil 1.9.200310 
 - tool-openocd 2.1000.200630 (10.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio/build/genericSTM32F103C8/firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  20.9% (used 4280 bytes from 20480 bytes)
Flash: [===       ]  30.6% (used 20064 bytes from 65536 bytes)
.pio/build/genericSTM32F103C8/firmware.elf  :
section            size        addr
.text             14724   134225920
.text.align           4   134240644
.ARM.exidx            8   134240648
.data              3208   536870912
.rodata            2120   134243864
.bss               1072   536874120
.debug_frame       1000           0
.comment            126           0
.ARM.attributes      41           0
Total             22303
<lambda>(["upload"], [".pio/build/genericSTM32F103C8/firmware.bin"])
AVAILABLE: blackmagic, dfu, jlink, serial, stlink
CURRENT: upload_protocol = dfu
MethodWrapper(["upload"], [".pio/build/genericSTM32F103C8/firmware.bin"])
Auto-detected: /dev/ttyACM0
maple_upload ttyACM0 2 1EAF:0003 "/home/tyler/Launchpad/Projects/AlcoholDispenser/.pio/build/genericSTM32F103C8/firmware.bin"
No DFU capable USB device found
dfu-util 0.7

Copyright 2005-2008 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2012 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Filter on vendor = 0x1eaf product = 0x0003
Waiting for /dev/ttyACM0 serial...Done
======================================================= [SUCCESS] Took 6.48 seconds =======================================================
```

If I open a serial:
```
Congratulations, you have installed the STM32duino bootloader
See https://github.com/rogerclarkmelbourne/STM32duino-bootloader

For more information about Arduino on STM32
See https://www.stm32duino.com
```

Any clue as to why I can't upload?

To me the discrepancy between dfu-util version 0.7 and 1.9.2 stands out in the logs, but I am not sure if that is the problem.