platformio/platform-ststm32 #208: genericSTM32F103C8: Undefined reference to 'end' symbol causes crash at malloc

maxgerhardt, 2019-02-28:
The following `platformio.ini` is used:

```
[env:bp_debug]
platform = https://github.com/platformio/platform-ststm32.git
board = genericSTM32F103C8
framework = arduino
monitor_speed = 4800
debug_tool = stlink
upload_protocol = stlink
lib_archive = false 
build_flags = -g3 -ggdb
```

Code used:

```cpp
#include <Arduino.h>

String i="";
void setup()
{
  i = "ABC";
}
void loop()
{
}
```

The code will crash the processor in the constructor of the `String` class when trying to execute a `realloc()`. When compiling, the following warning occurs: 

```
arm-none-eabi-g++ -o .pioenvs\bp_debug\firmware.elf -T jtag_c8.ld -Os -mthumb -mcpu=cortex-m3 -Wl,--check-sections -Wl,--gc-sections -Wl,--unresolved-symbols=report-all -Wl,--warn-common -Wl,--warn-section-align -Wl,--warn-unresolved-symbols .pioenvs\bp_debug\src\main.cpp.o -LC:\Users\Maxi\.platformio\platforms\ststm32\ldscripts -L.pioenvs\bp_debug -LC:\Users\Maxi\.platformio\packages\framework-arduinoststm32-maple\STM32F1\variants\generic_stm32f103c\ld -Wl,--start-group .pioenvs\bp_debug\libFrameworkArduinoVariant.a .pioenvs\bp_debug\libFrameworkArduino.a -lm -lgcc -Wl,--end-group
c:/users/maxi/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7-m\libnosys.a(sbrk.o): In function `_sbrk':
sbrk.c:(.text._sbrk+0x18): warning: undefined reference to `end'
arm-none-eabi-objcopy -O binary .pioenvs\bp_debug\firmware.elf .pioenvs\bp_debug\firmware.bin
MethodWrapper(["checkprogsize"], [".pioenvs\bp_debug\firmware.elf"])
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [==        ]  20.8% (used 4256 bytes from 20480 bytes)
PROGRAM: [===       ]  31.8% (used 20840 bytes from 65536 bytes)
```

The `_sbrk` , responsible for memory management in dynamic allocations, can't find the `end` symbol for the end of the heap. This is turn because the linker file doesn't declare it, it only declares `_end`. 

In `jtag_c8.ld`: `INCLUDE common.inc`. Then in `common.inc`

```
    .bss :
      {
        . = ALIGN(8);
        __bss_start__ = .;
        *(.bss .bss.* .gnu.linkonce.b.*)
        *(COMMON)
        . = ALIGN (8);
        __bss_end__ = .;
        _end = __bss_end__;
      } > REGION_BSS
```

Should be 

```
    .bss :
      {
        . = ALIGN(8);
        __bss_start__ = .;
        *(.bss .bss.* .gnu.linkonce.b.*)
        *(COMMON)
        . = ALIGN (8);
        __bss_end__ = .;
        _end = __bss_end__;
		end = _end;
      } > REGION_BSS
```

Either that or PlatformIO uses the wrong linker script for the `_sbrk()` implementation of the arduino-maple framework.

maxgerhardt, 2019-02-28:
Also see https://community.platformio.org/t/program-stopped-when-instantiating-string-class-plate-stm32f103c8-with-arduino-framework-in-vscode/6784

valeros, 2019-02-28:
Hi @maxgerhardt ! Many thanks for pointing this out. I see you used `
lib_archive = false` but this only applies to the official core. But anyway you are right, and we should link Arduino framework as object files so all symbols will be exported and used in the linking process. In the next package, `lib_archive = false` will be enabled by default. In case you want to try it now, open file`packages/framework-arduinoststm32-maple/tools/platformio-build-stm32f1.py`  and replace next lines in the build script:

```python
#
# Target: Build Core Library
#

libs = []

if "build.variant" in env.BoardConfig():
    env.Append(
        CPPPATH=[variant_dir]
    )
    libs.append(env.BuildLibrary(
        join("$BUILD_DIR", "FrameworkArduinoVariant"),
        variant_dir
    ))

libs.append(env.BuildLibrary(
    join("$BUILD_DIR", "FrameworkArduino"),
    join(FRAMEWORK_DIR, "cores", "maple")
))

env.Prepend(LIBS=libs)
```

with this:
```python
#
# Target: Build Core Library
#

if "build.variant" in env.BoardConfig():
    env.Append(
        CPPPATH=[variant_dir]
    )

    env.BuildSources(
        join("$BUILD_DIR", "FrameworkArduinoVariant"),
        variant_dir)

env.BuildSources(
    join("$BUILD_DIR", "FrameworkArduino"),
    join(FRAMEWORK_DIR, "cores", "maple"))
```

maxgerhardt, 2019-03-01:
@valeros Uhm actually `lib_archive = false` was kind of in there by accident and should have nothing to do with the error I think. I only tried this because intermediately I was getting a `arm-none-eabi-ar not found` error which was solved by un- and re-installing ststm32 platforms. The same error still occurs with this `platformio.ini`:

```
[env:bp_debug]
platform = https://github.com/platformio/platform-ststm32.git
board = genericSTM32F103CB
framework = arduino
``` 

Still leads to 

```
c:/users/maxi/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7-m\libnosys.a(sbrk.o): In function `_sbrk':
sbrk.c:(.text._sbrk+0x18): warning: undefined reference to `end'
```

Or did I not understand you?

valeros, 2019-03-03:
@maxgerhardt You right that the problem is in an incorrect version of `sbrk`. Currently, `sbrk` from the standard library is used, but we need `sbrk` that is implemented in the framework. To achieve that we link the framework in the form of object files. It should be fixed in the latest package, could you please update all packages for ststm32 and try again? Would be great if you upload build log somewhere in case your project fails to compile. Thanks!

maxgerhardt, 2019-03-04:
@valeros Thanks I will retry with your changes above because there are no new commits to this repository or the platformio-core repo relating to this.

maxgerhardt, 2019-03-05:
@valeros Okay this was fixed in the newest dev version without the need for me to update scriptfiles. Now correctly links and uses the correct `syscalls.c` with which the error is gone. Closing

closed 2019-03-05 by maxgerhardt