platformio/platform-ststm32 #213: Macros created from mbed_app.json are not conveyed in IDE

maxgerhardt, 2019-03-04:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** (`platformio --version`): `PlatformIO, version 4.0.0a3`

### Description of problem

When compiling a project that uses the `mbed_app.json`, the macros created from it are not showing up in the IDE integration. In my particular example, I'm compiling https://github.com/maxgerhardt/pio-lorawan-example, export the project for Eclipse. The project will compile without errors but Eclipse cannot resolve the symbols which should have been generated by the `mbed_app.json`. When using the `mbed-cli`, the `mbed_config.h` file would have these macro values.

#### Steps to Reproduce

1. `git clone https://github.com/maxgerhardt/pio-lorawan-example.git`
2. `cd pio-lorawan-example && pio init --ide=eclipse`
3. Open project with Eclipse
4. Observe unknown symbols in `main.cpp`, e.g. `MBED_CONF_LORA_DUTY_CYCLE_ON` which is define in the `mbed_app.json` as `"lora.duty-cycle-on": true`

### Actual Results

[image]

### Expected Results

No errors for resolving the symbol.

valeros, 2019-03-04:
Hi @maxgerhardt ! I'd say that's expected behavior as those macro are located in `mbed_config.h` and this file should be explicitly included. Could you please try to add the next two includes to your `main.cpp`:
```
#include <mbed.h>
#include <mbed_config.h>
```

maxgerhardt, 2019-03-04:
Okay that actually but worked but this code compiled correctly before meaning that all macros should have  be defined and included beforehand; strange.. I noticed that `mbed_config.h` lies in `.pioenvs/<target>/mbed_config.h` so this file only exists after you build once? This might explain why the C/C++ indexer can't find it, the file simply doesn't exist when the project is first created but not yet compield.

ivankravets, 2019-07-18:
See https://github.com/platformio/platform-ststm32/issues/213#issuecomment-469409901

closed 2019-07-18 by ivankravets