platformio/platform-ststm32 #214: PIO doesn't export toolchain includes to IDE project

maxgerhardt, 2019-03-04:
I have noticed that suddenly my IDEs (CLion and Eclipse) cannot resolve the toolchain's standard include files anymore. Things like `#include <stdio.h>` are not resolvable. 

This does not happen in e.g. a ESP8266 Arduino project genereted by PIO. For such a project, the include paths include e.g. 

```
${USERPROFILE}\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include
${USERPROFILE}\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include
```

etc in which these headers are found. However, for STM32 mbed projects, this doesn't happen. There is no reference to a `toolchain-gccarmnoneeabi` folder. 

Either the process by which these include paths are discovered via invoking the compiler is broken or PIO suddenly doesn't export these paths anymore.

PIO version is the latest dev version (`pio upgrade --dev`) and this was tested with the latest ststm32 platform (`platform = https://github.com/platformio/platform-ststm32.git`). As a sample project one can e.g. again take my https://github.com/maxgerhardt/pio-lorawan-example, do a `pio init --ide=eclipse` and see the results.

valeros, 2019-03-04:
Hmm, couldn't reproduce the issue with the latest Eclipse and toolchain paths are still in `.settings/language.settings.xml`
[image]

maxgerhardt, 2019-03-05:
@valeros Then I will upgrade my Eclipse and retry, thanks.

maxgerhardt, 2019-03-05:
@valeros You are right it does work with the latest `2018-12 (4.10.0)` version of Eclipse-CPP. Closing.

closed 2019-03-05 by maxgerhardt

maxgerhardt, 2019-03-05:
Also works for CLion in the newest version (2018.3.4)